adafruit/circuitpython #6259: Implement watchdog RAISE mode on RP2040

chabala, 2022-04-06:
The [watchdog module](https://docs.circuitpython.org/en/latest/shared-bindings/watchdog/) has two modes, `RAISE` and `RESET`. `RAISE` is to raise a `WatchDogTimeout` exception when the timer expires, while `RESET` causes a hard reset of the board.

Currently on RP2040, [RAISE mode is not implemented](https://github.com/adafruit/circuitpython/blob/b0bb5ff5d3df793794963b8eadbf2b888d51d0a7/ports/raspberrypi/common-hal/watchdog/WatchDogTimer.c#L77). Additionally, because `RESET` is the only available mode, [there is no way to deinit the watchdog when it is no longer needed](https://github.com/adafruit/circuitpython/blob/b0bb5ff5d3df793794963b8eadbf2b888d51d0a7/ports/raspberrypi/common-hal/watchdog/WatchDogTimer.c#L41). This makes using the watchdog to guard specific misbehaving areas of code impractical.

felker, 2023-03-13:
Is there still no way to deinit the watchdog? I ended up resorting to using this kludge when temporarily disabling the watchdog in my code

```
# from microcontroller import watchdog as w
# from watchdog import WatchDogMode

class Nop(object):
    def nop(*args, **kw): pass
    def __getattr__(self, _): return self.nop
w = Nop()
```

microdev1, 2023-03-13:
Hi, the following PR adds the functionality to `deinit` the watchdog and also reset it on `VM` exit. I am not sure if `RAISE` mode can be implemented as I don't think there is a way to subscribe a callback to the watchdog trigger event.
- #7718