adafruit/circuitpython #6260: Compiler doesn't like some CFLAGS in env

muranyia, 2022-04-06:
### CircuitPython version

```python
NA
```

### Code/REPL

```python
export CFLAGS=-march=native
make BOARD=xxxx
```

### Behavior

```make
arm-none-eabi-gcc: error: unrecognized -march target: native
arm-none-eabi-gcc: note: valid arguments are: armv4 armv4t armv5t armv5te armv5tej armv6 armv6j armv6k armv6z armv6kz armv6zk armv6t2 armv6-m armv6s-m armv7 armv7-a armv7ve armv7-r armv7-m armv7e-m armv8-a armv8.1-a armv8.2-a armv8.3-a armv8.4-a armv8.5-a armv8-m.base armv8-m.main armv8-r iwmmxt iwmmxt2
arm-none-eabi-gcc: error: missing argument to '-march='
```

### Description

It seems that the compiler cannot handle the `-march=native` option in CFLAGS.
IMO it would be best completely ignored.

dhalbert, 2022-04-06:
This is dependent on the compiler you are using. What version are you using? I'm not sure why you're doing this -- what's your use case? We wouldn't want to ignore the flag, we'd want to report the error.

If you would like to change build options, that's best done in the appropriate Makefile or .mk file.

dhalbert, 2022-04-06:
More:

`-march=native` is for the architecture the compiler is running on, but we are specifically using `arm-none-eabi-gcc` as a cross-compiler. If you want to make a native build, check out the `ports/unix` tree.

closed 2022-04-06 by dhalbert