adafruit/circuitpython #6264: ESP32-S2 TFT    TFT Power not on by default

kmelden, 2022-04-08:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-beta
Adafruit CircuitPython 7.2.0
Adafruit CircuitPython 7.2.4
Adafruit CircuitPython 7.2.5
```

### Code/REPL

```python
import digitalio
import board
import time

# Turn on I2C power pin
i2c_power = digitalio.DigitalInOut(board.I2C_TFT_POWER)
time.sleep(0.1)
i2c_power.switch_to_output()
time.sleep(0.1)
i2c_power.value = False
time.sleep(1)
i2c_power.value = True
```

### Behavior

TFT should by default and echo REPL.  It does not, except in 
CircuitPython 7.1.1  Note that the Backlight is on.

TFT is working when "double click" switch to get loader
and after a ROM reload.  Once any CircuitPython other
than 7,1.1 is loaded, the TFT is off by default.  I have checked
every version after 7.1.1, including both 7.3.0 BETAs

This code turns TFT OFF then ON and works only on 
CircuitPython 7.1.1

It does not work on:
Adafruit CircuitPython 7.3.0-beta
Adafruit CircuitPython 7.2.0
Adafruit CircuitPython 7.2.4
Adafruit CircuitPython 7.2.5

thanks
k

jfurcean, 2022-04-08:
It was changed from `I2C_TFT_POWER` to `TFT_I2C_POWER` in #6045 to match the other TFT pins.

kmelden, 2022-04-08:
I check pin name in REPL on each rev before running TFT code to turn TFT on and change code to match. In REPL run “import board” then “dir(board)” to get pin name.

dhalbert, 2022-04-08:
I just tried this in REPL on 7.2.4 on a Feather ESP32-S2 TFT:
```py
Adafruit CircuitPython 7.2.4 on 2022-03-31; Adafruit Feather ESP32-S2 TFT with ESP32S2
>>> import board
>>> board.TFT_I2C_POWER
board.TFT_I2C_POWER
>>> import digitalio
>>> p = digitalio.DigitalInOut(board.TFT_I2C_POWER)
>>> p.switch_to_output()   # turned the display OFF (default output is False)
>>> p.value = True   # turned the display ON
>>> p.value = False # turned the display OFF
>>> p.value = True  # turned the display back ON
```

Could you try it in the REPL?

ladyada, 2022-04-08:
@dhalbert the TFT/Repl should be on by default on boot - it used to like other TFT-native devices like the funhouse. i think scott changed some things with pin resets - maybe thats why it no longer works?

kmelden, 2022-04-08:
Adafruit CircuitPython 7.2.4 on 2022-04-06; Adafruit Feather ESP32-S2 TFT with ESP32S2
Board ID:adafruit_feather_esp32s2_tft

Did not work and TFT off by default still.
Works in 7.1.1 with board.I2S_TFT_POWER as pin name
I do not think that name of pin is the problem. I check it before
running the sample (I had a similar code snippet).

I think that I might need to do some reset in code ?
I know that most of that stuff is in C low level stuff and
probably needs lots of cleanup and inits.

Low power mode is important and maybe something for
that goal might have messed up the default TFT power ?
I did notice that when I ran p.value =True before changing 
that pin to an Output that it squawked that it was an INPUT.
thanks
k

dhalbert, 2022-04-08:
@kemelden Did you test in the REPL, as I did, or in a code.py? If the latter, could you show your code?

kmelden, 2022-04-08:
I ran that code both in REPL and in Mu framework with a 'code.py' file.

kmelden, 2022-04-08:
Both do not have TFT on by default or after running that code in Mu.
This is revisions 7.2.0 7.2.3 7.2.4 7.2.5 that do not work.
Revision 7.1.1 does work and TFT is on by default and
can be controlled on or off.  I have FLASHED the ROM a few
times and loaded CIrcuitPython and reset the board 5-10 times
under each revision.  Only 7.1.1 works.

dhalbert, 2022-04-08:
On my ESP32 TFT it works, and it is on by default. I am inquiring about whether there was an engineering change at some point. To be clear, the board you have is an actual https://www.adafruit.com/product/5300, and is not made from a regular ESP32-S2 Feather and a separate display? Is that right?

kmelden, 2022-04-08:
HI
It is Adafruit ESP32-S2 TFT Feather - 4MB Flash, 2MB PSRAM, STEMMA QT
Product ID: 5300   Integrated unit.  (I did notice that there did not seem
to be a board revision number on the board)

dhalbert, 2022-04-08:
OK, @ladyada's sample does not show the REPL, like yours, but mine works. What is in your `INFO_UF2.TXT` when the bootloader is running? Mine is:
```
TinyUF2 Bootloader cd9d938 - tinyusb (0.10.1-109-g9b3ec69b)
Model: Adafruit Feather ESP32-S2 TFT
Board-ID: ESP32S2-FeatherTFT-revA
Date: Dec 17 2021
```
And if you double-click to get the bootloader, do you see a colorful boot screen, like this?
[image]

kmelden, 2022-04-08:
TinyUF2 Bootloader cd9d938 - tinyusb (0.10.1-109-g9b3ec69b)
Model: Adafruit Feather ESP32-S2 TFT
Board-ID: ESP32S2-FeatherTFT-revA
Date: Dec 17 2021

dhalbert, 2022-04-09:
We are checking to see i there's a marginal timing power-up issue. We have one sample that acts like yours. Most do not, so it's been harder to reproduce.

kmelden, 2022-04-09:
Hi Dan
Thanks for update. 
I will run 7.1.1 for now as it works perfectly and I am just using gas sensor.
Kurt

dhalbert, 2022-04-10:
#6272 fixes this, but not in the way we'd expect, as it duplicates what the code in `espressif_board_reset_pin_number()` was trying to do. There may be a timing issue, or there may be some logic issue. Leave this open to diagnose further.

edmundopl, 2022-04-24:
Same problem here, on 2 different new boards, same boot loader info, same model

jwfmcclain, 2022-04-27:
FWIW, I seems to have the same problem, let me know if there is anything I can do to help test.

dhalbert, 2022-04-27:
> FWIW, I seems to have the same problem, let me know if there is anything I can do to help test.

Which version are you using? #6272 was merged late on April 9, so builds since then should work, such as 7.3.0-beta.2, released last night.

dhalbert, 2022-04-27:
@edmundopl Which version does not work for you? Try 7.3.0-beta.3 if you are further back.

jwfmcclain, 2022-04-28:
@dhalbert, just loaded 7.3.0-beta.2, and it does work, but it sounds like you aren't completely happy with the fix, so let know if there are any other builds you want tested.

kmelden, 2022-04-28:
7.3.0-beta.2 (and most, but not all, builds after 7.3.0-beta.1) seems to work after multiple installs and reboots. 

Thanks

tewner, 2022-05-01:
I had the same problem with a brand new ESP32-S2 TFT. 7.3.0-Beta cleared up the issue.

MartinJoeW, 2022-05-23:
> I had the same problem with a brand new **ESP32-S2 TFT. 7.3.0-Beta.2 cleared up the issue.**

FWIW this solve the issue for me as well.