adafruit/circuitpython #6271: ICM20948 IMU NOT FOUND by QTPY RP2040/Teensy Micromod/ESP32S3 running ICM20xx library

mjs513, 2022-04-10:
### CircuitPython version

```python
1. Adafruit CircuitPython 7.3.0-beta.1 on 2022-04-07; Adafruit QT Py RP2040 with rp2040
2. Adafruit CircuitPython 7.3.0-beta.0 on 2022-04-01; ESP32-S3-DevKitC-1-N8 with ESP32S3
3. Adafruit CircuitPython 5.3.0-rc.0-1823-gef9f9c8bf-dirty on 2022-04-07; Teensy MM with IMXRT1062DVL6A  (this is funny because it was built with later download afte 7.3.0beta.0
```

### Code/REPL

```python
ICM20048 code
import time
import board
import adafruit_icm20x
import busio

i2c = busio.I2C(board.SCL, board.SDA)  # or board.SCL1 and board.SDA1 for RP2040
icm = adafruit_icm20x.ICM20948(i2c)

while True:
    print("Acceleration: X:%.2f, Y: %.2f, Z: %.2f m/s^2" % (icm.acceleration))
    print("Gyro X:%.2f, Y: %.2f, Z: %.2f rads/s" % (icm.gyro))
    print("Magnetometer X:%.2f, Y: %.2f, Z: %.2f uT" % (icm.magnetic))
    print("")
    time.sleep(0.5)

SCANNER CODE:
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
#
# SPDX-License-Identifier: MIT

"""CircuitPython I2C Device Address Scan"""
# If you run this and it seems to hang, try manually unlocking
# your I2C bus from the REPL with
#  >>> import board
#  >>> board.I2C().unlock()

import time
import board
import busio

# To use default I2C bus (most boards)
i2c = busio.I2C(board.SCL, board.SDA)
# i2c1 = busio.I2C(board.SCL1, board.SDA1)
#i2c2 = busio.I2C(board.SCL2, board.SDA2)
# To use default I2C bus (most boards) scl/sda use for ESP32s3
#i2c = busio.I2C(board.IO10, board.IO11)
#i2c1 = busio.I2C(board.IO1, board.IO2)
# use bitbangio for BNO080 to be recognized
#i2c1 = bitbangio.I2C(board.I2C_SCL1, board.I2C_SDA1, timeout = 500)
#i2c1 = busio.I2C(board.I2C_SCL1, board.I2C_SDA1)
# for QT PY RP2040
# i2c = busio.I2C(board.SCL1, board.SDA1)

def i2cportscan():
    print("Wire")
    # Lock the I2C device before we try to scan
    while not i2c.try_lock():
        pass
    # Print the addresses found once
    # print(
    #    "I2C addresses found:", [hex(device_address) for device_address in i2c.scan()]
    # )
    for device_address in i2c.scan():
        print("Device found at address", end=" ")
        print(hex(device_address), end=" (")
        printKnownChips(hex(device_address))
        print(end=")")
[... 311 lines cut ...]
  File "adafruit_icm20x.py", line 679, in __init__
  File "adafruit_icm20x.py", line 224, in __init__
ValueError: No I2C device at address: 0x69

Code done running.
===============================================================

Wire
Device found at address 0x69 (MPU6050,MPU9050,MPU9250,ITG3701,L3G4200D )

Code done running.
  File "adafruit_icm20x.py", line 679, in __init__
  File "adafruit_icm20x.py", line 224, in __init__
ValueError: No I2C device at address: 0x69

Code done running.

### Description

- Error when using I2C
- Device found using I2C scanner at 0x69 using RP2040 and Teensy MicroMod
- Device not found by I2C scanner using a ESP32S3
- All 3 boards fail to find ICM20948 using the ICM20xx library.
- Hardware: sparkfun ICM20948
Messages from scanner and icm script:

mjs513, 2022-04-11:
In case anybody was looking I traced the problem to the QWIIC connector on the Breakout board.  I added pins to the board and the used a QWIIC connector to dupont wires - I connected the duponts on the breakout board and attached them to the stemma/qwiic connectors on the RP2040/TMM/T4.1 and then it started working.  Strange

jepler, 2022-04-11:
Closing since it sounds like you found a hardware problem. Please re-open or file a fresh issue if needed.

closed 2022-04-11 by jepler