adafruit/circuitpython #6274: ESP32-S2 TFT Feather needs static IP

ClarkInAz, 2022-04-12:
There needs to be a way to set a static IP, Gateway, and Netmask for the "native" WiFi on this board.

FWIW, I can get it to work on SparkFun's ESP32 MicroMod board (though admittedly it's an S3, not an S2) under the Arduino IDE.

Please see https://forums.adafruit.com/viewtopic.php?f=60&t=190084

anecdata, 2022-04-12:
This should (hopefully) be a straightforward implementation in espressif/common-hal:
https://github.com/adafruit/circuitpython/blob/main/ports/espressif/common-hal/wifi/Radio.c
using `esp_netif_set_ip_info()`:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/network/esp_netif.html#_CPPv421esp_netif_set_ip_infoP11esp_netif_tPK19esp_netif_ip_info_t
and feeding it the station netif or the AP netif from here:
https://github.com/adafruit/circuitpython/blob/7be66a57332fe63f6d2759663bf309b59162633d/ports/espressif/common-hal/wifi/Radio.h#L42
I won't have time to do this for some weeks, but if you'd like to try a PR, there are plenty of folks who could help, review, test, etc.

gamblor21, 2022-06-03:
Closing this issue as it was added in #6441

closed 2022-06-03 by gamblor21