adafruit/circuitpython #6285: Odd behavior with QT Py ESP32-S2 and CP 7.2.5

dekeyrej, 2022-04-15:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.5 on 2022-04-06; Adafruit QT Py ESP32S2 with ESP32S2
Board ID:adafruit_qtpy_esp32s2
```

### Code/REPL

```python
import gc, os

# Show available memory
print("Memory Info - gc.mem_free()")
print("---------------------------")
print("{} Bytes\n".format(gc.mem_free()))

flash = os.statvfs('/')
flash_size = flash[0] * flash[2]
flash_free = flash[0] * flash[3]
# Show flash size
print("Flash - os.statvfs('/')")
print("---------------------------")
print("Size: {} Bytes\nFree: {} Bytes\n".format(flash_size, flash_free))
```

### Behavior

Memory Info - gc.mem_free()
---------------------------
2043120 Bytes

Flash - os.statvfs('/')
---------------------------
Size: 963072 Bytes
Free: 854528 Bytes

### Description

ESPTOOL reports 4MB Flash.  941K available with code.  Perhaps the partitioning is off?

Neradoc, 2022-04-15:
That's the expected behaviour, CP sits in a 1.4MB partition, another is kept free for possible OTA updates.
Here is the partition map: <https://github.com/adafruit/tinyuf2/blob/master/ports/espressif/partitions-4MB.csv>

dekeyrej, 2022-04-15:
OK.  Glad to hear I didn't get a bad board.  Good that the behavior is "expected".

Do you suppose it would be possible to change the partition map to reallocate ota_1 to ffat?

closed 2022-04-15 by dekeyrej

dekeyrej, 2022-04-15:
Sorry for my less than appreciative previous response, Neradoc.  

Thank you for your prompt and informative answer to my post!