adafruit/circuitpython #6286: busio.I2C(SCL, SDA) giving "Attempted heap allocation when VM not running." on Save

phantomsixthplayer, 2022-04-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.5 on 2022-04-06; Adafruit Metro M4 Express with samd51j19
adafruit-circuitpython-metro_m4_express-en_US-7.2.5
```

### Code/REPL

```python
import busio
from board import *

with busio.I2C(SCL, SDA) as i2c:
    device = I2CDevice(i2c, self.address)
    with device:
        device.write(bytes_to_write)
```

### Behavior

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Attempted heap allocation when VM not running.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

### Description

When I click save in Mu randomly I will get the message heap allocation when VM not running.

I have a [Tic 36v4](https://www.pololu.com/product/3141) connected to GND, SDA and SCL pins on the Metro M4 Express.

phantomsixthplayer, 2022-04-20:
busio was having a default timeout for clock stretching of 255 microseconds but cannot change it, so instead we discovered that bigbangio accepts the timeout parameter. Since TIC max clock stretching is 1.5ms we set the timeout to 1500us.

dhalbert, 2022-04-21:
This is still a bug, so let's leave it open for diagnosis.

jepler, 2022-05-03:
I'm trying to duplicate this, but my set-up isn't like the original reporter, and the original report is not a complete program.

My test board is a pygamer, because it was handy, the same microcontroller family, and has a built-in i2c device

My test:
```python
import busio
from board import *
import adafruit_lis3dh 

i2c = busio.I2C(SCL, SDA)
assert i2c.try_lock()
print(i2c.scan())
i2c.unlock()
lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, address=25) # address jumper is pulled high on pygamer

while True:
    print(lis3dh.acceleration)
```

On my host computer, I run a loop which is intended to simulate writing a file from my editor every 3 seconds:
```
$ while sleep 3; do touch /media/jepler/CIRCUITPY/test.file; sync; done
```

my testing was done with a slightly modified version of 7.3.0 beta, though nothing relevant should be different:
> Adafruit CircuitPython 7.3.0-beta.2-8-gb292a925d on 2022-05-03; Adafruit PyGamer with samd51j19

phantomsixthplayer, 2022-05-07:
was having multiple issues at the time and wasn't able to get i2c working with the tic driver using busio, then switched to bitbangio as I could set the clock stretch to the same as the tic which is 1500us and haven't had issues since.

I'm having a feeling the heap allocation error might be a coincident when it crashed with using about 20 labels in conjunction with rgbmatrix, mp3decoder and neopixels. I did experience this a lot with the program before however I changed the design and only have the allocation error every so often when resaving via Mu.

dhalbert, 2022-05-08:
Thanks for the reply. I thought that the SAMD51 had a much longer clock stretching timeout than 255us. That was a problem on the RP2040, which we fixed a while ago. Please open another issue if that is the case. I will close this issue for now as it seems to not necessarily be reproducible.

closed 2022-05-08 by dhalbert