adafruit/circuitpython #6304: Issue with 'mpy-cross/Makefile.static-mingw'

gvanem, 2022-04-22:
### CircuitPython version

```python
From tip of `master` yesterday.
```

### Code/REPL

```python
None.
```

### Behavior

Building the `mpy-cross` program using MinGW, fails in the last link-step:
```
LINK mpy-cross.static.exe
x86_64-w64-mingw32-gcc -o mpy-cross.static.exe
...

x86_64-w64-mingw32-gcc.exe: error: f:gvVC_2019VCToolsMSVC14.29.30133libx86: No such file or directory
/usr/bin/sh: f:gvVC_2019VCToolsMSVC14.29.30133AtlMfclibx86: command not found
/usr/bin/sh: f:ProgramFiler-x86WindowsKitsLib10.0.22000.0umx86: command not found
```

Since the `$(LIB)` env-variable is used for the link. But this is a MSVC thing and it contains `\` slashes.
Easy fix for me was to just unset it first:
```diff
--- a/mpy-cross/Makefile.static-mingw 2022-04-21 08:04:09
+++ b/mpy-cross/Makefile.static-mingw 2022-04-22 07:52:25
@@ -6,5 +6,6 @@
 CROSS_COMPILE = x86_64-w64-mingw32-
 BUILD=build-static-mingw
 STATIC_BUILD=1
+LIB=

 include mpy-cross.mk
```

### Description

I've said enough.

### Additional information

None.

dhalbert, 2022-04-22:
> From tip of `master` yesterday.

Did you mean `main`? `master` is defunct.

We do build on Windows as part of our CI checks with MSYS, see here: https://github.com/adafruit/circuitpython/blob/main/.github/workflows/ports_windows.yml
Is there something different with what you are doing?

gvanem, 2022-04-22:
> We do build on Windows as part of our CI checks with MSYS, 

MSYS does not normally use (or need) a `LIB` env-var, does it? Like I wrote `LIB` is a MSVC (+Borland, Watcom) thing.

gvanem, 2022-04-22:
Try adding a `MAKEFLAGS += --warn-undefined-variables` to `py/mkenv.mk`. 
Then watch all the warnings. In a *CI* job a `warning: undefined variable 'LIB'` would probably show up.
But not here; it's a legitimate env-var for other compilers.