adafruit/circuitpython #6306: AiThinker ESP32-S3-32S dev kit I2C OLED issue

ITstreet1, 2022-04-23:
### CircuitPython version

```python
CircuitPython 7.2.0 on AiThinker ESP32-S3-32S dev kit
https://circuitpython.org/board/espressif_esp32s3_devkitc_1_n8/
https://docs.ai-thinker.com/_media/esp32/docs/nodemcu-esp-s3-32s-kit_v1.0.0specification.pdf

Using Thonny
```

### Code/REPL

```python
import board
import displayio
import adafruit_displayio_ssd1306
i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_addredd=0x3C)
```

### Behavior

>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 20, in <module>
AttributeError: 'module' object has no attribute 'I2C'
>>> 

### Description

I have in lib folder:
adafruit_bus_device folder
adafruit_displayio_ssd1306.mpy

### Additional information

I tried 7.2.5 but the board couldn't connect as CIRCUITPY.

Any info/help would be nice.

Neradoc, 2022-04-23:
`board.I2C()` is a shortcut for the default I2C pins on the board, as defined usually by the silkscreen.
That board does not have default I2C pins, so you gotta chose your pins and use busio.
For example if you were using IO1 and IO2:
```py
import board
import busio
i2c = busio.I2C(scl=board.IO1, sda=board.IO2)
```

Neradoc, 2022-04-23:
Also, I believe we don't have a build of Circuitpython for that AI Thinker S3, you are using the build for this Espressif board ?
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

ITstreet1, 2022-04-23:
Nice one.
But, here, there is no mention of this you typed:
https://learn.adafruit.com/monochrome-oled-breakouts/circuitpython-usage

By importing busio and with help(busio) I can see I2C class. This should be written on the page I mention.

Yes I know this is not for the AiThinker board. This is the closest to this dev kit. If you need me to try something for you with this board, please do.

I will try OLED and let you know if it works.

ITstreet1, 2022-04-23:
I2C SSD1306 OLED works like a charm with AiThinker ESP32-S3-32S dev kit and espressif_esp32s3_devkitc_1_n8 7.2.0 BIN.

ITstreet1, 2022-04-24:
I didn't find a function for screen rotation, font size, etc. Can someone put some light here?

Thank you.

jepler, 2022-04-24:
For 'how do I' questions, please use our forums or discord, thanks!

closed 2022-04-24 by jepler