adafruit/circuitpython #6307: pimoroni servo 2040: board.NEOPIXEL

jepler, 2022-04-23:
The board module has `LED_DATA` but it should probably be using the standard name `NEOPIXEL`.

ladyada, 2022-04-24:
@ZodiusInfuser

ZodiusInfuser, 2022-04-25:
If `board.NEOPIXEL` is the convention, and the appropriate name for multiple LEDs (rather than `board.NEOPIXELS`), then this sounds fine to change. I went with LED_DATA because it's more generic and matches the pin name on the schematic. This by extension means our C++, MicroPython, and CircuitPython board definitions are consistent.

Let me know whichever way, as we'll have other board hopefully launching this week with a similar setup that I'm yet to raise the PR for.

ladyada, 2022-04-26:
@ZodiusInfuser you can add an alias for NEOPIXEL - please submit a PR :)

jepler, 2022-04-26:
CircuitPlayground (Express & Bluefruit), PyGamer, and PyBadge at least have multiple neopixels attached to board.NEOPIXEL, that's not a problem.

PR #6339 (2022-05-03): Add NEOPIXEL pin
Closes: #6307
commit 7d7986b0 (2022-05-03): Add NEOPIXEL pin

Closes: #6307
closed 2022-05-03 by dhalbert