adafruit/circuitpython #6309: Teensy 4.x - Can the File system be preserved when you reprogram the board?

KurtE, 2022-04-25:
As I mentioned in the comments for #6287,  the latest version of the Teensy 4.x bootloader chips is now setup to not erase the full flash chip, but instead will always erase the first 512kb for normal Teensy boards or 1MB if the teensy board was purchased with the ability to download encrypted images and have the board locked such that only the person with the lock key file, can reprogram it.
Note: If your program is > 512KB the program code will erase more 64KB chunks.

Also note: that running the Teensy program that comes with the current release, will automatically update the T4.x boards to the new bootloader code. 

More details are up on the PJRC website page: https://www.pjrc.com/teensy/td_code_security.html

There is also some interesting information up on the PRJC forum thread: https://forum.pjrc.com/threads/67232-Teensy-4-What-triggers-Halfkay-to-do-a-full-flash-erase?p=304702&viewfull=1#post304702

That is today with Arduino code, one can do something like:

```
#include <LittleFS.h>
LittleFS_Program myfs;
...
void setup()
{

  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    // wait for serial port to connect.
  }
  if (!myfs.begin(15*1024*1024)) {
    Serial.printf("Error starting %s\n", "PROGRAM FLASH DISK");
    while (1) {
      // Error, so don't do anything more - stay stuck here
    }
  }
  Serial.println("LittleFS initialized.");
...

``
On a Teensy MicroMod and have a 15mb LittleFS survive having the program updated.

KurtE, 2022-04-29:
> I mentioned this somewhere else, but can't remember where: an alternative is to use TinyUF2 as an alternate bootloader. This is what we do on ESP boards to make loading CircuitPython easier for most uses.

Sorry, I don't understand enough about your internals for some of these things.  However with the Teensy boards, the bootloader support is built into hardware chip...   But again how it works on normal power up cycle or reboots without reprogramming your code is in control

dhalbert, 2022-04-29:
ESP32-S2 boards (and STM) also have a built-in bootloader. The TinyUF2 bootloader runs as a secondary bootloader on boards like that (the original bootloader always runs the secondary bootloader). A UF2 bootloader presents as a fake USB drive. You have probably seen this on other Adafruit boards, or on the RP2040. No other program is needed: you just drag the firmware .UF2 file to the drive to upload it.

KurtE, 2022-05-09:
@dhalbert  and others - I have been sort of curious why the Filesystem is wiped clean.  So I have been doing some experiments.   

Most of these are talked about on the Teensy form thread, that @roberthh (MicroPython) created for same issue.

https://forum.pjrc.com/threads/67232-Teensy-4-What-triggers-Halfkay-to-do-a-full-flash-erase?p=280127&viewfull=1#post280127

I hacked up a quick and dirty hex file range dump program from the Teensy_loader_cli app, that simply dumped the ranges of memory:
```
kurte@DESKTOP-BVOH5B0:~/github/teensy_loader_cli$ ./teensy_loader_dmp /mnt/d/firmware_41.hex 
60000000 - 60000200
60001000 - 60001030
6000c000 - 60010000
60010000 - 60020000
60020000 - 60030000
60030000 - 60040000
60040000 - 60050000
60050000 - 60060000
60060000 - 60070000
60070000 - 60080000
60080000 - 6008968c
$$
```
Confirmed that more or less same holes in address as MicroPython.

I also confirmed that it was erasing your logical record 1 (0 you have hard coded). 
More details in thread.

Today I wondered maybe it is the holes in address.  So I hacked up a copy of the teensy_loader_cli sketch to add a new option -f (fill the holes), which after it read in the hex file, it updates all of the areas between min/max that does not have any data defined....

I tried it and sure enough it appears like it now retained the file system... Only some simple tests so far.
I pushed up copies of these two programs up on new fork/branch: https://github.com/KurtE/teensy_loader_cli/tree/experiments

So now wondering how hard would it be to update these linker scripts to NOT put these holes in the address?

tannewt, 2022-05-09:
This sounds like a Teensy bootloader quirk more than an issue with our linker script. (Maybe the size detection code just assumes continuous ranges.) However, I wouldn't mind changing the linker script to take up continuous ranges if it means it preserves the filesystem.

I think you just need to modify the LENGTH parameters here for the first few ranges: https://github.com/adafruit/circuitpython/blob/main/ports/mimxrt10xx/linking/common.ld#L20=

tannewt, 2022-05-23:
You should be able to use the same flash config. I redid it to work with nvm.toml and the other iMX boards.

KurtE, 2022-05-24:
I now have what I believe is a working version.  So far I have tested it with T4.1 and Sparkfun_teensy_micromod.

will test tomorrow with T4.

Code is up in the branch: https://github.com/KurtE/circuitpython/tree/teensy_retain_fs

PR #6425 (2022-05-24): Teensy Builds retain Filesystem
As mentioned in the issue #6309,  The newer Teensy boot loader can retain a file system that is stored in the flash memory as long as it is not in the area that is automatically erased. 

Some of this is described on the web page: https://www.pjrc.com/teensy/td_code_security.html

But to do this, the bootloader code looks at specific things in the uploaded data to determine if it will retain any data or not and how much.

The two things I fixed was the Loader is looking for the value:  0x432000D1 at 0x60001000, it was 0x412000D1 

And the bootloader needs to kno [... cut]
closed 2022-05-24 by tannewt