adafruit/circuitpython #6313: Huffman submodule contents not included when cloning adafruit / circuitpython repo

CedarGroveStudios, 2022-04-26:
### CircuitPython version

```python
CircuitPython 7.2.5
```

### Code/REPL

```python
git clone https://github.com/adafruit/circuitpython.git
cd circuitpython/mpy-cross
make
```

### Behavior

```py
Traceback (most recent call last):
  File "../py/makeqstrdata.py", line 29, in <module>
    import huffman
ModuleNotFoundError: No module named 'huffman'
make: *** [build/genhdr/qstrdefs.enum.h] Error 1
make: *** Deleting file `build/genhdr/qstrdefs.enum.h'

### Description

After cloning `https://github.com/adafruit/circuitpython.git` for use in building `mpy-cross` (per https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/creating-a-library#mpy-3106477-16), making `mpy-cross` exits with this error:
```py
Traceback (most recent call last):
  File "../py/makeqstrdata.py", line 29, in <module>
    import huffman
ModuleNotFoundError: No module named 'huffman'
make: *** [build/genhdr/qstrdefs.enum.h] Error 1
make: *** Deleting file `build/genhdr/qstrdefs.enum.h'
```

The `circuitpython/tools/huffman` folder appears in the local cloned repo, but is empty.

### Additional information

`git submodule status` includes this entry:
`-27b1bba76198a0b343f694a6d680b5293d1c56aa tools/huffman`

deshipu, 2022-04-26:
Please refer to https://learn.adafruit.com/building-circuitpython/build-circuitpython on how to build CircuitPython. In this case you are missing `make fetch-submodules`.

dhalbert, 2022-04-26:
`huffman` is a Python module you need to install via `pip3 install requirements-dev.txt`. But @deshipu is right: you need to follow the recipe in https://learn.adafruit.com/building-circuitpython/build-circuitpython, as there are a number of steps to prepare your environment. The steps have changed over the years, including recently.

jepler, 2022-04-26:
Hijacking this a little bit, we have `tools/huffman` as a git submodule but also list `huffman` in `requirements-dev.txt`. We should do one or the other, but not both...

CedarGroveStudios, 2022-04-26:
Thank you. It’s difficult to find the correct cookbook to follow.

closed 2022-04-26 by CedarGroveStudios