adafruit/circuitpython #6318: ESP32-S2 pulseio Buffer over-read

mwisslead, 2022-04-27:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.5 on 2022-04-06; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
import board
import pulseio
import array
import time
x = pulseio.PulseIn(board.D5, maxlen=11)
y = pulseio.PulseOut(board.D6, duty_cycle=65535)
y.send(array.array('H', [350] * 23))
time.sleep(0.1)
print(x.popleft())
```

### Behavior

prints 0

### Description

A wire is connecting D5 to D6. The code should print a value around 350 but instead it prints 0. Looking at ports/espressif/common-hal/pulseio/PulseIn.c it appears that on line 51 and 57 the pulseio_pulsein_obj_t start value is incremented without bound once maxlen is reached and then used as an index to the buffer when popleft is called. Increasing the number of pulses sent makes it print other values than 0 but rarely anything around 350.

dhalbert, 2022-04-27:
Thank you for this report and diagnosis. This may be related to #4300 and perhaps some other `PulseIn` issues as well.

If you see a fix to prevent the overrun, a PR would be welcome.

mwisslead, 2022-04-27:
This issue is specific to popleft so shouldn't affect #4300 which is calling getitem.

It looks like this bug is duplicated across all ports that have pulseio except raspberrypi which is using a different convention to address the buffer.

It's a simple fix so I'll go ahead and submit a PR.

jepler, 2022-07-06:
Closed based on the above merged PR. Feel free to open a new issue or re-open this issue if there are still problems.

closed 2022-07-06 by jepler