adafruit/circuitpython #6322: Protect secrets.py file by making them unreadable

crazyiop, 2022-04-28:
Hi, I love the circuitpython on wifi capable board but this is always bugging me:
- Every tutorial tell you to put your wifi credentials/token in a secret.py files (to understandably  not version it with the code)
- Some board are placed just out of reach for other people to mess with, or at least plug a usb cable if they really want to.
- The mass storage function is amazing to upload new code but makes the secrets.py file extremely easily accessible too.

It would be nice to have a way to put secrets data into a board and having them stay a bit more secret.

Maybe not the best idea, but I have the following suggestion: 
When reading the content of a 'protected file' have it just return garbage, fixed or empty content. This include read from the usb mass storage but also from ampy when relevant.
The 'protected files' would be declared as such by their name or location. Some possible case are:
 - secrets.py (by default as it is the one already used in many places)
 - read_protected_*.py
 - /read_protected/*.py

It might not be feasible to hide those secrets to someone capable of dumping the memory, but it would considerably raise the bar for exploit as again that bar is currently as low as "plug an usb cable"...

deshipu, 2022-04-28:
You can always compile the file with mpy-cross to secrets.mpy, would that work for you?

crazyiop, 2022-04-28:
I have not thought of that as I have not played with mpy files yet. It would fix my main concern yes. Thank you very much !!

closed 2022-04-28 by crazyiop

dhalbert, 2022-04-28:
If someone opens the .mpy file with an editor, the strings are still readable, surrounded by binary stuff. It is not encrypted in any way. But you can raise the bar a bit by encoding the passwords and decoding them in your code.

You can disable making CIRCUITPY visible, though physical access to the hardware can override that:
https://learn.adafruit.com/customizing-usb-devices-in-circuitpython/circuitpy-midi-serial#circuitpy-mass-storage-device-3096583-4

crazyiop, 2022-04-29:
@dhalbert I had second thought and was planning to try later to inspect the mpy for that. Thanks for the clarification. Also thanks for this doc link, that seems a good option to me.