adafruit/circuitpython #6326: Documentation: Add more links for built in modules

jepler, 2022-04-30:
The CircuitPython core documentation is a terse reference, not a tutorial. In many cases, we have some other sources of documentation that we hope users will find useful. Let's link to them! I think most modules fall into one or more of these categories:
 * There's a tutorial guide for it. Link to the tutorial.
 * It's intended to be used by a wrapper library (for example, neopixel_write). Link to the wrapper library and to its learn guide
 * It's a subset of a standard Python module. Link to the Python docs with a standard disclaimer that the version for CircuitPython is intended to be a compatible subset except as noted.
Sometimes, this might mean adding multiple links (displayio, for example). In some cases, of course, none of these options may make sense. This could be because the library is suitably obscure (e.g., `_eve`) or for some other reason.

This will probably take some time to implement, so while the following checklist may be handy, it'd be good to double-check all modules before ultimately closing this issue.

- [ ] __future__
- [ ] _bleio
- [ ] _eve
- [ ] _pew
- [ ] _stage
- [ ] adafruit_bus_device
- [ ] adafruit_pixelbuf
- [ ] aesio
- [ ] alarm
- [ ] analogio
- [ ] atexit
- [ ] audiobusio
- [ ] audiocore
- [ ] audioio
- [ ] audiomixer
- [ ] audiomp3
- [ ] audiopwmio
- [ ] bindings
- [ ] bindings
- [ ] bindings
- [ ] bindings
- [ ] bitbangio
- [ ] bitmaptools
- [ ] bitops
- [ ] board
- [ ] busio
- [ ] camera
- [ ] canio
- [ ] circuitpython_typing
- [ ] countio
- [ ] digitalio
- [ ] displayio
- [ ] dualbank
- [ ] espidf
- [ ] espidf
- [ ] floppyio
- [ ] fontio
- [ ] framebufferio
- [ ] frequencyio
- [ ] gamepadshift
- [ ] getpass
- [ ] gifio
- [ ] gnss
- [ ] i2cperipheral
- [ ] imagecapture
- [ ] ipaddress
- [ ] is31fl3741
- [ ] keypad
- [ ] math
- [ ] mdns
- [ ] memorymonitor
- [ ] microcontroller
- [ ] msgpack
- [ ] multiterminal
- [ ] neopixel_write
- [ ] nvm
- [ ] onewireio
- [ ] os
- [ ] paralleldisplay
- [ ] ps2io
- [ ] pulseio
- [ ] pwmio
- [ ] qrio
- [ ] rainbowio
- [ ] random
- [ ] rgbmatrix
- [ ] rotaryio
- [ ] rp2pio
- [ ] rp2pio
- [ ] rtc
- [ ] samd
- [ ] samd
- [ ] sdcardio
- [ ] sdioio
- [ ] shared-bindings/
- [ ] sharpdisplay
- [ ] socketpool
- [ ] ssl
- [ ] storage
- [ ] struct
- [ ] supervisor
- [ ] synthio
- [ ] terminalio
- [ ] time
- [ ] touchio
- [ ] traceback
- [ ] uheap
- [ ] ulab
- [ ] usb
- [ ] usb_cdc
- [ ] usb_hid
- [ ] usb_host
- [ ] usb_midi
- [ ] ustack
- [ ] vectorio
- [ ] videocore
- [ ] watchdog
- [ ] wifi
- [ ] zlib

tekktrik, 2022-05-01:
Happy to help with this effort!

jepler, 2022-05-03:
I hoped it would catch your attention! :)

tekktrik, 2022-05-16:
Could the following modules be checked?

- `board`
- `busio`
- `bitbangio`
- `digitalio`
- `pwmio`
- `displayio`
- `storage`
- `neopixel_write`
- `analogio`
- `qrio`
- `fontio`
- `terminalio`
- `microcontroller`
- `audiomp3`
- `rp2pio`
- `alarm`
- `bitmaptools`

Thanks!

tekktrik, 2022-05-16:
Will do!

tekktrik, 2022-05-16:
Will do!

FoamyGuy, 2022-07-11:
In #6581 I've added links for:
- keypad
- sharpdisplay
- msgpack
- touchio

tannewt, 2022-08-11:
Doc fixes go live immediately so we don't need this in the 8.0.0 release.

ladyada-eagleclaw, 2026-10-02:
Marked only the `rgbmatrix` checklist entry complete: its module documentation already links the dedicated RGB LED matrix Learn guide.

https://github.com/adafruit/circuitpython/blob/6fefe633d1768436ca43c70008fc2ddce42dacdd/shared-bindings/rgbmatrix/__init__.c#L16

The remaining module checklist stays open.