adafruit/circuitpython #6335: QTPy ESP32-S3 worryingly high temp (150ºF) / high current (140mA) when idle

todbot, 2022-05-03:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-beta.2 on 2022-04-27; Adafruit QT Py ESP32-S3 no psram with ESP32S3
```

### Code/REPL

```python
import _bleio
```

### Behavior

I'm experiencing worryingly high-temps (150ºF) and high-current draw (140mA) on [QT Py ESP32-S3](https://circuitpython.org/board/adafruit_qtpy_esp32s3_nopsram/) when sitting idle in the REPL. Right out of reset it'll draw ~50mA but then jump to 100mA. Upon `import _bleio` current draw immediately goes to 140mA and keeps rising, with Vreg temperature feeling dangerously hot. 

### Description

This seems CircuitPython ESP32-S3 internals-related, but may be a general ESP32-S3 issue.

I'm experiencing worryingly high-temps (150ºF) and high-current draw (140mA) on [QT Py ESP32-S3](https://circuitpython.org/board/adafruit_qtpy_esp32s3_nopsram/) when sitting idle in the REPL. Right out of reset it'll draw ~50mA but then jump to 100mA. Upon `import _bleio` current draw immediately goes to 140mA and keeps rising, with Vreg temperature feeling dangerously hot. 

These below photos were taken about 10 minutes apart. There's a thermocouple kapton'd on top of the Vreg.  Running `Adafruit CircuitPython 7.3.0-beta.2 on 2022-04-27; Adafruit QT Py ESP32-S3 no psram with ESP32S3` and on a fresh local build.

On a [Feather ESP32-S3 w/ no PSRAM](https://circuitpython.org/board/adafruit_feather_esp32s3_nopsram/) and `Adafruit CircuitPython 7.3.0-beta.2 on 2022-04-27; Adafruit Feather ESP32S3 No PSRAM with ESP32S3` current draw is also 140mA on but the board doesn't get nearly as hot: about 100ºF at the Vreg near the BAT pin.

On a [QT Py ESP32-S2](https://circuitpython.org/board/adafruit_qtpy_esp32s2/), idle or connected to WiFi is around 50mA and the highest draw I can get is 100mA when running a [simple HTTP JSON API server using `ampule`](https://gist.github.com/todbot/8cc221d7b5aa89bf35fe3c641a740a32) and a simple hammer of it with `while [ 1 ] ; do ; curl http://192.168.42.174/status ; done`
Is the ESP32-S3 just a more power-hungry chip, is there some chip setting that's wrong, or is there an issue with the vregs?

[image]
[image]

dhalbert, 2022-05-03:
Filed on my request after Discord discussion. Thanks!

ladyada, 2022-05-03:
@brentru check yours? mine gets warm but not 150

todbot, 2022-05-03:
I was worried is a one-off issue two and checked the other QTPy ESP32-S3 that I have and both behave similarly.

ladyada, 2022-05-03:
what if you use arduino?

todbot, 2022-05-03:
I just tried the esp32 2.0.3-RC1 Arduino core.  And:

* 20mA - ROM bootloader mode
* 40mA - TinyUSB bootloader mode (not used but interesting)
* 100mA - Running `void loop() {Serial.println("Hello world!"); delay(500);}`
* 156mA - Running "BLE_uart" example but not connecting

So not a CircuitPython issue exactly but wth?

ladyada, 2022-05-03:
what about a wifiscan example?

todbot, 2022-05-03:
Good call. 

* 110mA - Running "WiFiScan" example Arduino sketch

Thermocouple on vreg reading about 112ºF.

ladyada, 2022-05-03:
note your temps should be in C - 150*C is silicon limit, do you mind converting/editing comments above?

todbot, 2022-05-03:
Done. Apologies for that.

ladyada, 2022-05-03:
ok just to verify you're getting the same power draw on the Feather and the QT Py S3, its just the S3 is warmer right? 
there's not a lot of ground plane on the QT to dissipate the generated current, and it _is_ a dual core 240 mhz processor. so being toasty isnt that out of bounds

@UnexpectedMaker are you seeing similar power draws? i dont have one of your boards with me right now to check :)

UnexpectedMaker, 2022-05-03:
@ladyada I see approx 145mA draw when importing `_bleio` - but temp wise on my TinyS3, it's not getting above 45deg C after about 5 mins.
[image]

dhalbert, 2022-05-03:
I am also seeing similar power consumption, but not excessively high temperatures. The regulator on top is at most 50C, and the ESP32-S3 is about 45C. I have essentially the same measuring equipment as @UnexpectedMaker. I also tested with a Thermoworks Thermapen Mk4 cooking thermometer, which has more sinking ability.

My QT Py has header pins installed, which are getting warm and probably acting as a good heat sink. @todbot, it appears you do not have header pins installed.

ladyada, 2022-05-03:
@todbot assuming the current is OK (seems like its high for just an import but like, not a hardware issue) - can you check the temp rise of the other 2x S3 QTPys u have? it may be our RF isnt tuned well enough and/or im wondering if its a one-off

todbot, 2022-05-03:
I'm thinking this is just a vreg heatsinking issue with the QTPy just being so small that there's not much copper on the board and the ESP32 being known for having 130mA BLE "by default" that people complain about it. 

The Feather ESP32-S3 I have has similar current consumption, but its vregs don't get that hot (40ºC at most)

I tested all 3 of my QT Py ESP32-S3s:

* All three behaved the same: 140mA when running "BLE_uart" Arduino sketch
* Headers do seem to act as heatsinks, it was 55ºC. (Pic 1)
* The no-header one was 60ºC in a vice (vice as heat-sink?) 
* After 20 mins, the free-floating with headers one went to 170mA and 60ºC (Pic 2)

Ambient was 23ºC, 43% humidity, no fans. 

Janky test setup for reference:

"Pic1" - Both headers & no-headers in a vice after 5 minutes
[image]

"Pic2" - headers free-floating after 20 mins
[image]

UnexpectedMaker, 2022-05-03:
@todbot I think it's less about the LDO and more about the lack of ways for heat to exit the board. My TinyX boards run hotter than my Feather format boards, with the same specs. The Feather boards have 3x more PCB surface exposure to air, allowing more heat to get into the copper pours, and then dissipate out off the board.

The QT Py is smaller than my TinyX boards and double sided as well, so even less air contact area for heat to dissipate. That might be reflected in components that get hot, like the LDO and the ESP32, but 170mA is not taxing the LDO on the QT Py much. It can do 500mA sustained and peak to 600mA.

todbot, 2022-05-03:
I was assuming it was the LDO producing most of the heat, but yeah totally. From reading, looks like the ESP32 heats up too.

brentru, 2022-05-03:
Tested: Qt Py ESP32-S3 running WipperSnapper Sketch (1x AHT20 sending measurements to AIO every 5m), arduino-esp32 2.0.3-RC1 core.

Avg. QT Py ESP32-S3 Temp (NO headers): 24degC
Avg. _Environmental_ Temp/Humid: 18degC/48%

ladyada, 2022-05-03:
@todbot yah your boards probably have less-than-ideal tuning which causes some RF self-heating, but also - its a small board, with a powerful chip. the regulator is doing a lot and there isn't a ton of space for it. your board isnt damaged and wont be damaged from use, its just a toasty lil boy :)

dhalbert, 2022-05-03:
@todbot We'll close this for now, and note the temperature characteristics in the guide. We appreciate your original writeup: it was definitely worth investigating, especially since it might have been a CircuitPython issue. Thanks.

closed 2022-05-03 by dhalbert

ladyada, 2022-05-03:
@todbot you could attach a smol heatsink if you like
https://www.adafruit.com/product/1493

todbot, 2022-05-03:
Agree on closing this issue. Everything seems to be working as intended. I was all excited to try BLE on ESP32 and was startled it was so hot when idly grabbing it. But after reading up, I see ESP32 BLE is much less developed than NRF. Not surprising, I guess.

@ladyada ahaha that heatsink is adorable! I'm sure I'd fat-finger it to oblivion. And right now I'd rather use an NRF52840-based approach.

stanelie, 2024-11-14:
I'm seeing 71ºC on several Waveshare ESP32-S3-Zero boards, running wifi only. I've tried lowering the CPU speed to 80000000, but I see no change in temperature. Very small boards.

dhalbert, 2024-11-14:
@stanelie as described above, the board itself does not provide much of a heatsink. You could add a heatsink if you are worried. There is not anything "wrong" -- the chip is consuming a noticeable amount of current which must be dissipated somehow. The recommended chip temperature is –40 to 105 C, so it's not outside the spec.