adafruit/circuitpython #6336: NeoPixel on ESP32-S3 QT Py not Working

MadGeometer, 2022-05-03:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.5 on 2022-04-06; Adafruit QT Py ESP32-S3 no psram with ESP32S3
Board ID:adafruit_qtpy_esp32s3_nopsram
```

### Code/REPL

```python
import time
import board
import neopixel

print("Hello, world!")

pixels = neopixel.NeoPixel(board.NEOPIXEL, 1)

pixels.brightness = 0.3

while True:
    pixels.fill((255, 0, 0))
    time.sleep(0.5)
    pixels.fill((0, 0, 0))
    time.sleep(0.5)
    print("Hello, world!!!")
```

### Behavior

soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello, world!
Hello, world!!!
Hello, world!!!

*** NeoPixel does not blink! ***

### Description

NeoPixel does not blink!

Note: this exact same code does work on ESP32-S2 QT Py.

MadGeometer, 2022-05-03:
This is also happening with 
adafruit-circuitpython-adafruit_qtpy_esp32s3_nopsram-en_US-7.3.0-beta.2.uf2

ladyada, 2022-05-03:
try absolute latest here
https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_qtpy_esp32s3_nopsram/en_US/

MadGeometer, 2022-05-03:
No, still doesn't work :-(

MadGeometer, 2022-05-03:
Sorry, my bad - I just installed the .uf2 file, not the .bin file! Here's the .bin version that is working:

Adafruit CircuitPython 7.3.0-beta.2-6-gbf0e1fafa on 2022-05-02; Adafruit QT Py ESP32-S3 no psram with ESP32S3
Board ID:adafruit_qtpy_esp32s3_nopsram

ladyada, 2022-05-03:
@todbot this is resolved right?

MadGeometer, 2022-05-03:
Yes, thank you!

jepler, 2022-05-03:
If a "uf2" and a "bin" behave differently, that is a bug that needs to be addressed.

todbot, 2022-05-03:
I just tried both the .bin and the .uf2 and both work as intended. I think this issue can be closed.

Pedantically, here's what I did:

### Test .BIN

- Put QTPy ESP32-S3 into JTAG/Serial bootloader mode (hold BOOT + click RESET)
- Go to https://nabucasa.github.io/esp-web-flasher/
  - Connect to QTPY serial port
  - Click "Erase", wait for it to complete
  - Load "adafruit-circuitpython-adafruit_qtpy_esp32s3_nopsram-en_US-20220502-bf0e1fa.bin"
   from "https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_qtpy_esp32s3_nopsram/en_US/"
  - Click "Program", wait for it to complete
- Press QTPy RESET button
- Observe CIRCUITPY drive appears
- Open REPL, and type:
    ```py 
    import board, neopixel; led = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.2); led.fill(0xff00ff)
    ```
- Observe onboard NeoPixel goes purple

### Test .UF2

- Put QTPy ESP32-S3 into JTAG/Serial bootloader mode (hold BOOT + click RESET)
- Go to https://nabucasa.github.io/esp-web-flasher/
  - Connect to QTPY serial port
  - Click "Erase", wait for it to complete
  - Load TinyUF2 "combined.bin" from "tinyuf2-adafruit_qtpy_esp32s3-0.9.0.zip" from "https://github.com/adafruit/tinyuf2/releases/tag/0.9.0"
  - Click "Program", wait for it to complete
- Press QTPy reset button
- Observe QTPYS3BOOT drive appears
- Copy to "QTPYS3BOOT" drive the file "adafruit-circuitpython-adafruit_qtpy_esp32s3_nopsram-en_US-20220502-bf0e1fa.uf2" from "https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_qtpy_esp32s3_nopsram/en_US/"
- Observe CIRCUITPY drive appears
- Open REPL, and type:
    ```py 
    import board, neopixel; led = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.2); led.fill(0xff00ff)
    ```
- Observe onboard NeoPixel goes purple

dhalbert, 2022-05-03:
Thank for retesting!

closed 2022-05-03 by dhalbert