adafruit/circuitpython #6348: rp2040 pulsio.PulseIn last pulse at beginning

mwisslead, 2022-05-04:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-beta.2 on 2022-04-27; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import time
import board
import pulseio

pulsein = pulseio.PulseIn(board.GP21, maxlen=5)
time.sleep(2.2)
pulsein.pause()
if len(pulsein) == 5:
    x = [pulsein.popleft() for _ in range(5)]
    print(x)
else:
    print(len(pulsein))
```

### Behavior

Output doesn't match input. Final pulse width is at the beginning of the list

### Description

Input pulses are  high, low, high, low, high of lengths [373, 215, 285, 327, 479] us with a 1 second period of low between them.

Output is [471, 376, 217, 287, 329]. The final value should be about 479 but it is at the beginning. This happens every time even though the input pulses are not synchronized with code execution. If the final pulse just wasn't counted, I would expect a value of 65535 at the beginning corresponding the the 1 second low period.

### Additional information

I looked at ports/raspberrypi/common-hal/pulseio/PulseIn.c and I can't identify the issue exactly but for some reason the start buffer index is reset to 0 in several places. I don't think that should be happening. There is an extra buf_index variable that is unnecessary and doesn't exist for other ports.

DavePutz, 2022-05-05:
@mwisslead , by design (which I am not particularly fond of), pulsein on all CP boards  wraps around if the incoming pulse is longer than the maxlen. I would prefer throwing a "Buffer Exceeded" or some such. In any case, it looks like your example is picking up an initial pulse (low?) then the five pulses (making six) and so wrapping the fifth around to the first position. You could work around this behavior by setting maxlen to 6 and dropping the first value.

mwisslead, 2022-05-05:
I believe it should wrap around in the internal buffer but the order of popleft should always be first in first out

mwisslead, 2022-05-05:
Switching the pulses to alternating 300us high 70000us low with the same code always gives an output of [65535, 300, 0, 0, 0]

DavePutz, 2022-05-05:
Since pulsein is a very primitive protocol, lacking an end of transmission indicator, the only way to 'detect' the end of a series of pulses is by timing out. The somewhat arbitrary timeout value on the RP2040 was set to 65535 ( a little more than 1 second), which is why you see that in your output. If you have a use case that needs a longer pulse length than that, please let's discuss it.
I will look at seeing if the order of pulses can be retained in the case of buffer overflow; but something is going to end up being overwritten (which is why I would prefer throwing an error).

dhalbert, 2022-05-05:
> by design (which I am not particularly fond of), pulsein on all CP boards wraps around if the incoming pulse is longer than the maxlen. I would prefer throwing a "Buffer Exceeded" or some such.

We would certainly be interested in adding this capability; it sounds like you have uses cases. It could be a new boolean option that defaults to the old way. It could be a constructor arg or a settable attribute or some other arg.

mwisslead, 2022-05-05:
@DavePutz are you saying that [65535, 300, 0, 0, 0] is what I should expect from alternating 300us high pulses and 70000us low pulses? I was expecting [65535, 300, 65535, 300, 65535] or [300, 65535, 300, 65535, 300]. Should I get [65535, 300, 0, 0, 0] using the 5 other ports that have pulseio? Can we agree that something is wrong if I run the same test with a esp32s2 and get a different result?

DavePutz, 2022-05-06:
The 70000us pulse will be interpreted as and end-of-pulse, since it exceeds 65535. So, yes, I would expect that to terminate a pulsein. What I am unsure of is that leading 65535. The default idle_state on a pulsein is false, meaning that no signal should be getting counted until it goes high. I am investigating that. As for the esp32s2, that is using a ring buffer provided by the FreeRTOS layer; and I am not sure what that uses to detect an end-of -pulse.

mwisslead, 2022-05-06:
Testing the esp32s2 with the same python code just different pin name, I get no pulses at all for alternating 300us high 70000. I upped the off pulse to 100000 and I get `[16128, 306, 303, 303, 303]` which except for the first value is ok if long idles are intended to be ignored. For the sequence of `[373, 215, 285, 327, 479]`, I get `[33444, 228, 297, 339, 483]` sometimes and `[0, 228, 297, 339, 483]` other times. If I drop the sleep time to 1.1 I get `[384, 228, 297, 339, 483]`.

mwisslead, 2022-05-07:
I figured out the issue is the start index not being updated so when you popleft or getitem you are getting values that aren't necessarily the first ones that came in.

PR #6361 (2022-05-08): Fix buffer issue for rp2040 PulseIn
Updates the starting index when maxlen is reached. Fixes #6348
closed 2022-05-08 by dhalbert