adafruit/circuitpython #6364: usb module isn't found in 7.3.0-beta.2

omijh, 2022-05-08:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-beta.2 on 2022-04-27; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
help('modules')
```

### Behavior

It will return all the modules without usb module 
```
__future__        board             onewireio         touchio
__main__          builtins          os                traceback
_asyncio          busio             paralleldisplay   ulab
_bleio            collections       pulseio           ulab
_eve              countio           pwmio             ulab.fft
adafruit_bus_device                 digitalio         qrio              ulab.linalg
adafruit_bus_device.i2c_device      displayio         rainbowio         ulab.numpy
adafruit_bus_device.spi_device      errno             random            ulab.scipy
adafruit_pixelbuf floppyio          re                ulab.scipy.linalg
aesio             fontio            rgbmatrix         ulab.scipy.optimize
alarm             framebufferio     rotaryio          ulab.scipy.signal
analogio          gc                rp2pio            ulab.scipy.special
array             getpass           rtc               ulab.utils
atexit            gifio             sdcardio          usb_cdc
audiobusio        imagecapture      select            usb_hid
audiocore         io                sharpdisplay      usb_midi
audiomixer        json              storage           uselect
audiomp3          keypad            struct            vectorio
audiopwmio        math              supervisor        watchdog
binascii          microcontroller   synthio           zlib
bitbangio         micropython       sys
bitmaptools       msgpack           terminalio
bitops            neopixel_write    time
Plus any modules on the filesystem
```

also if i `import usb` in my code it will return 
`no module named 'usb'`

### Description

Missing USB module but mentioned on the circuit python download page for 7.3.0-beta.2

jepler, 2022-05-08:
This module is not (yet) supported in any CircuitPython build, and it's a mistake that it appears in the documentation.

omijh, 2022-05-08:
I understand, Thank you. 
It's in documents & the download page 
https://docs.circuitpython.org/en/latest/shared-bindings/usb/index.html#module-usb
https://circuitpython.org/board/raspberry_pi_pico/

tannewt, 2022-05-10:
It is a bug that the `usb` module appears on the download page. It is only enabled on iMX RT ports currently. RP2 will be next once TinyUSB has PIO host support.

omijh, 2022-05-11:
Thank you, I am waiting for TinyUSB  host or USBHost library for sam in arduino to be ported too for my use case.

PR #6288 (2022-05-14): List frozen modules in the support matrix
This adds the frozen modules to the support matrix.
- each module is designated by its import name. eg: `neopixel` not `Adafruit_Circuitpython_Neopixel`
- frozen modules appear without a link, since those links are generated by sphinx as internal links
- the `usb` module is fixed to only show on boards with `CIRCUITPY_USB_HOST` (Closes: #6364)

Questions:
- Any idea how to make Sphinx link to the libraries RTD pages ?
- Should the frozen libraries be listed separately from the built-in ones or visually differentiated in some way ?

[image]
closed 2022-05-11 by omijh