adafruit/circuitpython #6367: pulsein / HC-SR04 regression in samd 7.x vs 6.x

ladyada, 2022-05-08:
### CircuitPython version

```python
for itsybitsy m0 and gemma m0

Your adafruit_hcsr04.mpy does not work with CircuitPython 7.x -- To get the sensor to work, I had to install:

adafruit-circuitpython-gemma_m0-en_US-6.3.0.uf2

and then use:

adafruit-circuitpython-bundle-6.x-mpy-20211213.zip

After downgrading the Gemma to these version - the sensor now works perfect.

I tried multiple versions of 7.X - none worked, it was only when I downgraded back to 6.3.0 - that the sensor started working.
```
```

### Code/REPL

```python
import time
import board
import adafruit_hcsr04
sonar = adafruit_hcsr04.HCSR04(trigger_pin=board.RX, echo_pin=board.TX)

while True:
    try:
        print((sonar.distance,))
    except RuntimeError:
        print("Retrying!")
    time.sleep(0.1)
```

### Behavior

```
(17.1204,)
(0.0,)
(0.0,)
(0.0,)
(0.0,)
(17.1204,)
(17.1204,)
(17.1204,)
```

Which is just random - no relation to distance of anything in front of the sensor. It's always just jumping between 0.0 and 17.xx

mwisslead, 2022-05-10:
Looks like @dhalbert removed pulseio for the itsybitsy m0 express in commit 843598ec. I tested this with an itsybitsy m0 express and rcwl-1601 and it does work but the accuracy is terrible without pulseio.

dhalbert, 2022-05-10:
We traded off `pulseio` against other things when we needed space. Perhaps the library should not attempt to work without `pulseio`.