adafruit/circuitpython #6369: `espressif_board_reset_pin_number()` is not working right for ESP32-S2 Feather TFT GPIO21 reset to high

dhalbert, 2022-05-09:
#6272 fixes #6264, but the code seems redundant: we thought `espressif_board_reset_pin_number()` code was supposed to be doing this reset to high already. Diagnose why `espressif_board_reset_pin_number()` is not producing the desired effect.

anecdata, 2022-07-29:
Is this related? I don't seem to be able to set up the pun to go low-power.

```
Adafruit CircuitPython 7.3.2 on 2022-07-20; Adafruit Feather ESP32-S2 TFT with ESP32S2
>>> import board, digitalio
>>> i2c_tft_power = digitalio.DigitalInOut(board.TFT_I2C_POWER)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: TFT_I2C_POWER in use
```

DJDevon3, 2022-07-29:
The ESP32-S2 feather guide https://learn.adafruit.com/adafruit-esp32-feather-v2/power-management says deep sleep is activated by turning NEOPIXEL_I2C_POWER pin low. On the ESP32-S2 TFT feather an additional regulator was added and should be triggered by turning off TFT_I2C_POWER which is traced back to pin GPIO21 on the espressif module. 

There is a difference how deep sleep is entered on the feather ESP32-S2 and feather ESP32-S2 TFT.
[image]

[image]

Hope that helps.

ladyada, 2022-08-01:
please note that turning off the regulator *does not trigger deep sleep mode* - only the deep sleep command does that. turning off the regulator *will reduce power greatly which will make entering deep sleep use the minimal power possible* (approx 100uA) 
@kattni plz check guides "The low power section is also missing for the S3 in its learn guide." etc

dhalbert, 2022-09-16:
There is a lot in this issue which is now addressed by #6907.

PR #6909 (2022-09-16): Remove unneeded workaround for Feather ESP32-S2 TFT
- Fixes #6369. Really finishes the fix, by removing the now unneeded workaround. The actual fix was due to #6898.

Tested on a Feather ESP32-S2 TFT. It no longer blanks the display on restart.
closed 2022-09-16 by MicroDev1