adafruit/circuitpython #6371: Xiao/Qt-py filter for downloads

alustig3, 2022-05-09:
A Xiao/QTPy checkbox under the filter->features section of the  https://circuitpython.org/downloads?sort-by=downloads would be useful.

alustig3, 2022-05-09:
also, the text for **"Wi-Fi (Two Version Onboard Antenna Or UFL Connector)"** seems to have been split across two checkboxes
[image]

jepler, 2022-05-12:
Thanks for the suggestion. I've fixed the second problem you noted, and offered a pull request to add the new 'feature', though I don't know whether Adafruit will choose to accept it.

If you have further issues or enhancements specifically about the circuitpython.org website, you can file them over on that repo: https://github.com/adafruit/circuitpython-org/ -- though it's no trouble that you filed this issue here.

Thanks again!

alustig3, 2022-05-12:
Thanks a lot!

closed 2022-05-12 by makermelissa

jepler, 2022-05-12:
Thanks again for the suggestion.