adafruit/circuitpython #6386: Investigate how best to run pre-commit

jepler, 2022-05-12:
I noticed an odd message from the pre-commit run in our github ci:
> [@octokit/rest] `const Octokit = require("@octokit/rest")` is deprecated. Use `const { Octokit } = require("@octokit/rest")` instead

While pursuing this, I found that the action we use (pre-commit/action) is deprecated. Pre-commit instead proposes you use their CI service, https://pre-commit.ci/

On the other hand, transitioning from the current action to a two-step shell command `pip install pre-commit; pre-commit run --all` would let us stop worrying about the deprecated action and not require us to begin using a new CI service.

Whatever we decide, we should do the same for the core and libs, and update cookiecutter to cut a library with whatever way we choose to invoke pre-commit.

dhalbert, 2022-05-12:
I read about this. For open-source repos, it is free, including for an organization. It will automatically fix things that can be fixed, which is good, but requires users to know to pull the automatic changes back when submitting a PR. It will be faster than doing `pip install pre-commit`, but that is not a big timesink in the scheme of things. The automatic fixing would save a lot of tiny commits for people who forget to install pre-commit locally.

jepler, 2022-05-12:
I tried out pre-commit.ci on two of my personal projects .. https://results.pre-commit.ci/run/github/77753953/1652368838.qvt34fwsRSiYze0oigbXMQ https://results.pre-commit.ci/repo/github/293359557

It's too bad their URLs are opaque. It does run fast, but in one of the two cases I ran into trouble where pre-commit behaved different than locally OR on github actions.

Here's an example of it pushing to a pull request: https://github.com/jepler/wwvbpy/pull/31 https://results.pre-commit.ci/run/github/77753953/1652369245._qJ40VmDTM2jffJpA_bWNQ