adafruit/circuitpython #6402: MemoryError MatrixPortal

Craigxyzzy, 2022-05-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-rc.0 on 2022-05-15; Adafruit Matrix Portal M4 with samd51j19
Board ID:matrixportal_m4
```

### Code/REPL

```python
from adafruit_matrixportal.matrixportal import MatrixPortal as mp
m = mp(width=64, height=64, bit_depth=6)

# following lines are NOT needed to produce error
# error occurs with or without attached matrix
m.set_background("/bats.bmp")
while True:
    continue
```

### Behavior

Traceback (most recent call last):
  File "code.py", line 2, in <module>
  File "adafruit_matrixportal/matrixportal.py", line 97, in __init__
  File "adafruit_matrixportal/graphics.py", line 67, in __init__
  File "adafruit_matrixportal/matrix.py", line 192, in __init__
MemoryError: 

Code done running.

### Description

Works fine with cp725, fails with 730rc0

### Additional information

MatrixPortal part 4745 & matrix part 3649
UF2 Bootloader v3.14.0 SFHWRO
Model: Matrix Portal M4
Board-ID: SAMD51J19A-MatrixPortal-v0

NINA firmware 174

dhalbert, 2022-05-16:
@makermelissa Do you know why this might happen?

@Craigxyzzy so just
```py
m = mp(width=64, height=64, bit_depth=6)
```
is enough to trigger it? Could you try some of the earlier 7.3.0 beta versions? Thanks.
https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/matrixportal_m4/en_US/

makermelissa, 2022-05-16:
Maybe try decreasing the bit depth. This will reduce the number of possible colors, but also decrease memory usage. More information is available here: https://learn.adafruit.com/rgb-led-matrices-matrix-panels-with-circuitpython.

Craigxyzzy, 2022-05-16:
m=mp(width=64,height=64,bit_depth=1) works 730b0
m=mp(width=64,height=64,bit_depth=1) works 730b1
m=mp(width=64,height=64,bit_depth=1) works 730b2
m=mp(width=64,height=64,bit_depth=1) works 730rc0

m=mp(width=64,height=64,bit_depth=2) fails 730b0
m=mp(width=64,height=64,bit_depth=2) fails 730b1
m=mp(width=64,height=64,bit_depth=2) fails 730b2
m=mp(width=64,height=64,bit_depth=2) fails 730rc0

image looks improper on 64x64 matrix:
m=mp(width=32,height=32,bit_depth=1) works 730rc0
m=mp(width=32,height=32,bit_depth=2) works 730rc0
m=mp(width=32,height=32,bit_depth=4) works 730rc0
m=mp(width=32,height=32,bit_depth=6) works 730rc0

works perfectly with 725, should also with 730...
m=mp(width=64,height=64,bit_depth=6) works 725

library files 2022-MAY-15
adafruit_bitmap_font
adafruit_display_text
adafruit_fakerequests.mpy
adafruit_io
adafruit_matrixportal
adafruit_minimqtt
adafruit_portalbase

makermelissa, 2022-05-16:
Yikes. Yeah, a bit_depth of 2 is very low and shouldn't be failing.

dhalbert, 2022-05-17:
This is failing really early, due to Protomatter (used by `RGBMatrix`) being unable to allocate a large enough chunk. It's not due to the libraries. Still investigating.

makermelissa, 2022-05-17:
I figured this was in the correct place. The libraries are only a wrapper for helping to initialize it.

dhalbert, 2022-05-18:
It appears that there is more fragmentation starting in 7.3.0 than in 7.2.x. It is not clear why, but we started 7.3.0 immediately with the MicroPython 1.18 merge, and a build of the PR merge commit shows the problem.

Craigxyzzy, 2022-05-18:
Success with cp730rc1 - Thank You @dhalbert @tannewt @makermelissa  Closing issue.

PR #6406 (2022-05-18): Restore automatic gc_collect() after an import
- Fixes #6402.

- The automatic garbage collection done after an `import` got lost in the MicroPython v1.18 merge, which rewrote the import code extensively. Restore it. This reduces heap fragmentation, which can make prevent allocating big buffers after significant imports.
- Improved error message for `RGBMatrix` being unable to allocate buffers. Previously, it just threw a blank `MemoryError`.
closed 2022-05-18 by Craigxyzzy