adafruit/circuitpython #6404: espressif: `re` runs out of stack

dhalbert, 2022-05-17:
The stack size on ESP32-xx may be too small. This code gets a stack overflow on an ESP32-S2 board running 7.2.5
```py
>>> import re
>>> _REQUEST_RE = re.compile(r"(\S+)\s+(\S+)\s")
>>> _REQUEST_RE.match("GET /" + ("a" * 135) + " whatev")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: maximum recursion depth exceeded
>>> _REQUEST_RE.match("GET /" + ("a" * 134) + " whatev")
<match num=3>
```
(originally noted in https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/pull/3)

RetiredWizard, 2022-05-17:
Just a thought....

I routinely compile custom images with MICROPY_ENABLE_PYSTACK turned off and MICROPY_STACKESS turned on. I don't know what I'm trading off with these settings but I've been assuming I've just been making the garbage collector's job harder and reducing the amount of memory I'm able to allocate in my applications.

If the advantage of using the stacks is to keep larger chunks of contiguous memory available, I wonder if they are still necessary on the new multi megabyte microcontroller boards.

dhalbert, 2022-05-18:
There is still a C stack that's running and is presumably used by the C-based `re` code. It think it may be that stack that is not big enough.

richardanaya, 2022-05-27:
I'm having difficulties doing 512-bit rsa key parsing as well.

richardanaya, 2022-05-27:
(https://gist.github.com/richardanaya/2158b09c52c3b2ea904aaeb48c524062

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 9, in <module>
  File "/lib/adafruit_jwt.py", line 101, in generate
RuntimeError: pystack exhausted
```)

RetiredWizard, 2022-05-28:
So I know building with PyStack=0 and Stackless=1 probably means you can't tell much from my results, but just in case it helps...
I though I'd try running @richardanaya 's code on my build. As soon as I try and import adafruit_jwt I get the following:

     Traceback (most recent call last):
       File "<stdin>", line 1, in <module>
     RuntimeError: maximum recursion depth exceeded

dhalbert, 2022-06-07:
There is a maximum task stack size which is fairly low at the moment (2k??)

dhalbert, 2022-06-21:
The size of the C stack for the main FreeRTOS task, `CONFIG_ESP_MAIN_TASK_STACK_SIZE`, is set to 8kB here:
https://github.com/adafruit/circuitpython/blob/d0953e8f845a7544a27bea92bbcd533f0a84d493/ports/espressif/esp-idf-config/sdkconfig.defaults#L297
(Confusingly, note there is also an `CONFIG_MAIN_TASK_STACK_SIZE`, whose value does not seem to affect this.)

If I change this to 16384, then I don't get the error above. The `re` library we are using uses recursion, and suffers from this stack problem. Here's an older bug from MicroPython, which was fixed by checking for stack overflow: https://github.com/micropython/micropython/issues/2451.

We could grow the stack; on other relatively RAM rich ports, we usually set the stack to 24kB, such as SAMD51 and nRF52480.

@tannewt I should discuss this with you. I'm confused about why `CIRCUITPY_DEFAULT_STACK_SIZE` does not affect this: I thought that determined the size of the C stack, but apparently not.

PR #6725 (2022-08-09): increase main task stack size from 8kB to 16kB
Fixes #6404.

Redo of #6514, which updated 7.3.x. This updates 8.0.0 builds.

- Increased default main task stack to 16kB in `sdkconfig.defaults.
- Removed stack size settings from sdkconfig-esp32.defaults`. Also removed some incorrect leftover debug UART settings from the legacy section of `sdkconfig-esp32.defaults`.

Tested wifi on Feather HUZZAH32 with basic internet test. There is some concern this might reduce the number of available sockets, so more testing by users is in order.
closed 2022-08-09 by tannewt