adafruit/circuitpython #6412: Unable to find 1.2" 7-segment I2C device when powered on 3.3v

makermelissa, 2022-05-19:
### CircuitPython version

```python
CircuitPython 6.0.0 beta 1 and later
```

### Code/REPL

```python
import board
from adafruit_ht16k33.segments import BigSeg7x4

i2c = board.I2C()
display = BigSeg7x4(i2c)
display.brightness = 0.5
display.print("ABCD")
```

### Behavior

Can't find the device when the I2C voltage is 3.3 and running CircuitPython 6.0.0 Beta 1 or later.

### Description

I have kind of a weird issue here. This is a follow up for guide feedback. The guide for https://www.adafruit.com/product/1270 was written before CP 5.0.0 was released, so I tested different versions and found it broke between 6.0.0 Beta 0 and 6.0.0 Beta 1. 

Note, if you up the I2C voltage to 5v (connected to the USB pin) it works, but this seems like it could damage the board.

### Additional information

I have it wired to a Metro M4 according to https://learn.adafruit.com/adafruit-led-backpack/circuitpython-wiring-and-setup-2666a265-2d8e-4cde-bcb8-dadc29b6b8fa.

dhalbert, 2022-05-20:
See my comment here: https://forums.adafruit.com/viewtopic.php?f=60&t=186294&hilit=segment&start=15#p903654 about my success fixing this for a previous user by using stronger pullups. As I think you know, this display has two power pins, one 5v for the display itself and for the controller, and one for the I2C voltage, which in this case is 3.3V.

makermelissa, 2022-05-20:
Ok, that's a good solution and one I can add to the guide. I'll verify it fixes it first.

makermelissa, 2022-05-23:
Closing because I added the proposed solution of the higher value pull-ups to the guide already.

closed 2022-05-23 by makermelissa