adafruit/circuitpython #6417: MAX4466 driver

Myslicki, 2022-05-21:
Driver support for the Electret Microphone Amplifier - MAX4466 with Adjustable Gain to work with circuit python.

Current setup is MAX4466  --->  adafruit voice bonnet --> raspberrypi W via the JST and pin connectors and would love to access the circuit python audio libraries with this microphone hardware.

ladyada, 2022-05-22:
what are you expecting to be able to do with the microphone?

Myslicki, 2022-05-24:
> what are you expecting to be able to do with the microphone?

audio signal processing of the sound for use in ML classification. Ideally can process the audio concurrently during data recording

dhalbert, 2025-02-01:
This is an analog amplifier. The only control available is a shutdown pin. So it doesn't really need a driver. General ADC audio in support will work with the output. On the RPi, you can use the Voice Bonnet guide for guidance.

closed 2025-02-01 by dhalbert