adafruit/circuitpython #6419: Check number of parameters: int.from_bytes()

rsbohn, 2022-05-22:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0-rc.2 on 2022-05-21; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
b = (0xFEED).to_bytes(2,"little")
# next line is not correct code
x = int.from_bytes(b, 2, "little")
print(x==0xFEED)
```

### Behavior

C Python raises a TypeError if too many arguments:

>>> int.from_bytes(bytes([0xFE,0xED]), 2, "big")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: from_bytes() takes exactly 2 positional arguments (3 given)

### Description

Expectation: Raise TypeError int.from_bytes() is called with more or less than 2 arguments.

rsbohn, 2022-05-23:
Maybe adding some examples to the documentation would be enough:

    int.from_bytes(b"\x00\x01","little") # returns 256
    int.from_bytes(b"\x00\x01","big") # returns 1
    
    (1).to_bytes(2,"little") # returns b'\x01\x00'
    (1).to_bytes(2,"big") # returns b'\x00\x01'

tannewt, 2022-05-23:
`from_bytes` source is here: https://github.com/adafruit/circuitpython/blob/0b58f1754d8f0cfbae82468138ced8f021e2ab0d/py/objint.c#L483

skerr92, 2022-10-16:
Appears to be fixed by #6562

dhalbert, 2022-10-16:
Thanks! Closing.

closed 2022-10-16 by dhalbert