adafruit/circuitpython #6429: ESP32 QT Py pin mapping broken in 7.3.0

alustig3, 2022-05-24:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit QT Py ESP32S2 with ESP32S2
Board ID:adafruit_qtpy_esp32s2
```

### Code/REPL

```python
import board
import rotaryio

class Encoder:
    def __init__(self,pin1,pin2):
        self.encoder = rotaryio.IncrementalEncoder(pin1, pin2)
        self.last_pos = 0

    def read(self):
        change = 0
        if self.encoder.position > self.last_pos:
            change = 1
        elif self.encoder.position < self.last_pos:
            change = -1
        self.last_pos = self.encoder.position
        return change

enc1 = Encoder(board.SCK, board.RX)
enc2 = Encoder(board.MOSI, board.MISO)

while True:
    a,b = enc1.read(),enc2.read()
    if a or b:
        print(a,b)
```

### Behavior

I have 2 rotary encoders connected the QTpy. The expected behavior of the above code is that when I rotate enc1 the first (left) value of the print statement shows a change, and when I rotate enc2, the second (right) value of the print statement shows a change. 

Curiously, when I upgraded from 7.2.4 to 7.3.0, the first encoder that is physically hooked up to SCK and RX does not produce any readings. The second encoder that is physically hooked up to MOSI and MISO, shows changes in the first(left) value of the print statement as if it were the first encoder hooked up up to SCK and RX.

So somehow MOSI and MISO got remapped to SCK and RX maybe? and I don't know what SCK and RX got remapped to.

the issue is resolved by switching back to 
```
Adafruit CircuitPython 7.2.4 on 2022-03-31; Adafruit QT Py ESP32S2 with ESP32S2
Board ID:adafruit_qtpy_esp32s2
```

dhalbert, 2022-05-24:
I tested and found that if you reverse the order of the creattion of `enc1` and `enc2`, then the the second one does not work: it doesn't seem to be the choice of pins.

There were recent changes to `rotaryio` since 7.2.x was branched, in #6207. @domdfcoding, tagging you for interest.

dhalbert, 2022-05-24:
I think the problem may be that `pcnt_unit_config()` is called twice in `ports/espressif/common-hal/rotaryio/IncrementalEncoder.c`, but it also finds a free unit each time it's called, so it's allocating two units. I have an idea for a fix.

dhalbert, 2022-05-25:
@alustig3 Builds that I think fix this problem are here: https://github.com/adafruit/circuitpython/actions/runs/2381784979. Scroll down a bit to see "Artifacts".

alustig3, 2022-05-25:
@dhalbert the new build worked for me. Thanks!

dhalbert, 2022-05-25:
Fixed by #6430.

closed 2022-05-25 by dhalbert