adafruit/circuitpython #6448: Dualbank not working ESP32-S2

calcut, 2022-05-31:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0 on 2022-05-23; FeatherS2 with ESP32S2
Board ID:unexpectedmaker_feathers2
```

### Code/REPL

```python
import adafruit_requests
import ssl
import socketpool
import wifi
import dualbank
from secrets import secrets

print('connecting to wifi')
wifi.radio.connect(secrets["ssid"], secrets["password"])
print('wifi connected')

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

url = 'https://downloads.circuitpython.org/bin/unexpectedmaker_feathers2/en_GB/adafruit-circuitpython-unexpectedmaker_feathers2-en_GB-7.3.0.bin'
response = requests.get(url)

chunk_size = 1024
offset=0
for chunk in response.iter_content(chunk_size):
    print(f'offset = {offset}')
    dualbank.flash(chunk, offset=offset) # <-- fails here with RuntimeError: Update failed
    offset += chunk_size

# # This doesn't work either
# image = response.content
# print('got content')
# dualbank.flash(image)   # <-- fails here with RuntimeError: Update failed
# print('flash complete')
```

### Behavior

Code successfully connects and can can receive the new firmware.
However dualbank.flash() just gives RuntimeError: Update failed

### Description

This happens whether I try to flash all at once, or do it in chunks.
Note I used a featherS2 board as it has enough RAM to not need to do chunks.
I have also tried on an Adafruit ESP32-S2 Feather.

I have also tried with
`storage.remount("/", readonly=False)` in boot.py, which didn't help

Relevant code seems to be here:
https://github.com/adafruit/circuitpython/blob/main/ports/espressif/common-hal/dualbank/__init__.c

calcut, 2022-05-31:
@MicroDev1

MicroDev1, 2022-05-31:
Hi, @calcut, the `bin` fetched from [circuipython.org](https://circuitpython.org/), starts at address `0x0` and includes various partitions that can't be updated using dualbank, the actual firmware is at `0x10000` in the `bin` (checkout the following partition table).
https://github.com/adafruit/circuitpython/blob/ebe442cd9dac31204f3d4f74ec615040ed21fe2a/ports/espressif/esp-idf-config/partitions-16MB.csv#L1-L10
I recommend fetching the firmware from [CircuitPython's S3 Bucket](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/) as it allows fetching in chunks and selective ranges:
```py
# Start Address 0x10000 (refer: partitions.csv)
header = {}
header["Range"] = "bytes=65536-"

url = 'https://adafruit-circuit-python.s3.amazonaws.com/bin/microdev_micro_s2/en_US/adafruit-circuitpython-microdev_micro_s2-en_US-7.3.0.bin'
response = requests.get(url, headers=header)
````
Above is for flashing the firmware in whole, for smaller chunks you can poll `S3` in a loop, each time updating the `header`.

Hope, I solved your issue, looking forward to your guide :)

closed 2022-05-31 by MicroDev1

calcut, 2022-06-06:
Thanks a lot for that, makes sense.
Unfortunately I'm still struggling to implement it. I'm pretty sure I had it work once, but can't figure out what I've changed to break it!

Any tips?

```
import adafruit_requests
import ssl
import socketpool
import wifi
import dualbank
from secrets import secrets

print('connecting to wifi')
wifi.radio.connect(secrets["ssid"], secrets["password"])
print('wifi connected')

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

# Start Address 0x10000 (refer: partitions.csv)
# https://github.com/adafruit/circuitpython/tree/main/ports/espressif/esp-idf-config

header = {}
header["Range"] = "bytes=65536-"
url = 'https://adafruit-circuit-python.s3.amazonaws.com/bin/unexpectedmaker_feathers2/en_GB/adafruit-circuitpython-unexpectedmaker_feathers2-en_GB-7.3.0.bin'

response = requests.get(url, headers=header)
print('got response')
dualbank.flash(response.content)   # <-- fails here with RuntimeError: Update failed
print('flash complete')
dualbank.switch()
```

```
code.py output:
connecting to wifi
wifi connected
got response
Traceback (most recent call last):
  File "code.py", line 24, in <module>
RuntimeError: Update failed
```

calcut, 2022-06-06:
for the record, I also tried this
`dualbank.flash(response.content, offset=65536)   # <-- fails here with RuntimeError: Update failed`

calcut, 2022-06-06:
Got it working on the Feather ESP32-S2 board (but not the FeatherS2, going back to that next)

```
import adafruit_requests
import ssl
import socketpool
import wifi
import dualbank
from secrets import secrets

print('connecting to wifi')
wifi.radio.connect(secrets["ssid"], secrets["password"])
print('wifi connected')

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

url = 'https://adafruit-circuit-python.s3.amazonaws.com/bin/adafruit_feather_esp32s2/en_GB/adafruit-circuitpython-adafruit_feather_esp32s2-en_GB-7.3.0.bin'

header = {}
offset=65536
position = 0
chunk_size=65536

length=1354336 #determined manually just now.

while position <= length:
    header["Range"] = f"bytes={position+offset}-{position+offset+chunk_size-1}"
    response = requests.get(url, headers=header)
    status = response.status_code
    print(f'{position=} {round(position/length*100, 1)}%')
    dualbank.flash(response.content, offset=position)
    position += chunk_size

print('flash complete')
dualbank.switch()
```

calcut, 2022-06-06:
OK, the same code but with updated url does not work on the FeatherS2

`https://adafruit-circuit-python.s3.amazonaws.com/bin/unexpectedmaker_feathers2/en_GB/adafruit-circuitpython-unexpectedmaker_feathers2-en_GB-7.3.0.bin`

I get `RuntimeError: Update failed` on the first iteration of the while loop.

I'm guessing because all the flash is external on that board??

calcut, 2022-06-06:
https://github.com/adafruit/circuitpython/issues/6448#issuecomment-1147321312

Sigh... this is not longer working for me. No idea what is going on!
I may come back to this another time!

calcut, 2022-06-06:
OK, the 2 questions I currently have:

1) Can Dualbank work with external Flash (as on FeatherS2)?

2) On Adafruit Feather ESP32-S2, I can flash one of the partitions successfully, but then if I reset and run the same code again it fails. To get it working again I have to run dualbank.switch() and hard reset again.
Does this mean dualbank can't overwrite the original boot partition? Or is there another explanation?

MicroDev1, 2022-06-08:
> Can Dualbank work with external Flash (as on FeatherS2)?

Yes, it can.

> ...Does this mean dualbank can't overwrite the original boot partition? Or is there another explanation?

Both `ota_0` (primary/original) or `ota_1` (secondary/alternate) can be overwritten from each other.

Multiple checks can result in the `Update Failed` error, the following check prevents duplicate firmware flashing:
https://github.com/adafruit/circuitpython/blob/ee3ccbc61b42e723cd92eba190d950dcefaf2507/ports/espressif/common-hal/dualbank/__init__.c#L86-L98
I believe this is the reason you are getting the error when running the code again. I feel `dualbank` should have more detailed and user-friendly error messages instead of a single generic one, this can be taken up as a separate issue.

calcut, 2022-06-15:
Nice one, thanks @MicroDev1 
"ESP_LOGW(TAG, "New version is the same as running version.")"
That was my problem.

Easy to get confused with the current generic message as you say.
I will aim to write something up about this over the next few weeks.

hgoldwire, 2023-10-25:
oops.  failure to RTMF.  

for others trying this,

daniel-alsen, 2025-05-02:
Great detail here guys!

Does dualbank.flash() now automatically detect which of the two OTA partitions is active and writes at the given offset from the start of the passive OTA partition?

The Dualbank documentation now says:

"Writes one of the two app partitions at the given offset."

Otherwise, I guess that you would have to detect which OTA partition is active and then adjust the offset based on that? As you will start at OTA_0, then run from OTA_1 until you update again, then run from OTA_0 again, etc.

Do I understand it right?

All the best,
Daniel