adafruit/circuitpython #6451: BLE disconnect event crash

callmeduncan, 2022-05-31:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit ItsyBitsy nRF52840 Express with nRF52840
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time

import adafruit_ble
from adafruit_ble_adafruit.adafruit_service import AdafruitServerAdvertisement
from adafruit_ble_adafruit.temperature_service import TemperatureService

# PyLint can't find BLERadio for some reason so special case it here.
ble = adafruit_ble.BLERadio()  # pylint: disable=no-member

connection = None

while True:
    print("Scanning for an Adafruit Server advertisement...")
    for adv in ble.start_scan(AdafruitServerAdvertisement, timeout=5):
        connection = ble.connect(adv)
        print("Connected")
        break

    # Stop scanning whether or not we are connected.
    ble.stop_scan()

    if connection and connection.connected:
        temp_service = connection[TemperatureService]
        while connection.connected:
            print("Temperature:", temp_service.temperature)
            time.sleep(1)
```

### Behavior

This is the 'ble_adafruit_simpletest_client.py' from 7.x bundle, I am also running 'ble_adafruit_simpletest.py' on a second board to connect with it.
If I cause the connection to drop by rebooting / interrupting the board running 'ble_adafruit_simpletest.py' the other board running 'ble_adafruit_simpletest_client.py' locks up.
REPL can no-longer be accessed
Mass storage is still working

### Description

This seems to affect all 7.x releases

On 6.3 it crashes in the same way if the board is powered on from cold, however if you power on, ctrl+c to the REPL, ctrl+d to resume and then disconnect the other side it doesn't crash, but it does not drop out of the `while connection.connected:` loop.

### Additional information

https://forums.adafruit.com/viewtopic.php?f=60&t=191584&p=926989
`supervisor.disable_ble_workflow()` does not seem to help

DavePutz, 2022-06-16:
Looks like CP is stuck in the 
```
    while (!read_info.done) {
        RUN_BACKGROUND_TASKS;
    }
```
loop in common_hal_bleio_gattc_read(). The BLE_GAP_EVT_DISCONNECTED event is not being handled in  _on_gattc_read_rsp_evt(), which means that read_info.done will never become true. I am not sure, however,
what is the right thing to do when the disconnect happens. I tested handling the disconnect and setting read_info.done
with a len of 0; but that just led to a TypeError in adafruit_ble/characteristics/float.py

tannewt, 2022-06-16:
Could you set done and then check if disconnected after the loop? If disconnected, then throw a ConnectionError. Thanks for looking into this @DavePutz

FoamyGuy, 2022-07-11:
I confirmed this does now raise the ConnectionError on `7.3.x` branch.

I tested with a Feather Bluefruit Sense and CLUE devices both using builds made from `7.3.x` branch. 

Output on the device running the _client.py code is now:
```
Traceback (most recent call last):
  File "code.py", line 28, in <module>
  File "adafruit_ble/characteristics/float.py", line 46, in __get__
  File "adafruit_ble/characteristics/__init__.py", line 246, in __get__
  File "adafruit_ble/characteristics/__init__.py", line 142, in __get__
ConnectionError: Not connected

Code done running.
```
I think this issue can be closed now, but wanted to check if anyone else knew of anything else still outstanding?

callmeduncan, 2022-07-11:
Thanks for sorting this!
Only thing that might still need to be done is the ble examples need updating to catch the ConnectionError condition.

closed 2022-07-20 by dhalbert