adafruit/circuitpython #6457: board.SPI() raises exception on Challenger NB RP2040 WiFi board

NathanY3G, 2022-06-03:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0 on 2022-05-23; Challenger NB RP2040 WiFi with rp2040
```

### Code/REPL

```python
>>> import board
>>> spi_bus = board.SPI()
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid pins

### Description

This exception also occurs when `busio` is used directly e.g.

```
busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid pins
```

### Additional information

The invalid pin appears to be related to MISO - it is possible to construct a `SPI` object for output only.

dhalbert, 2022-06-03:
@PontusO could you talk a look at this?

Neradoc, 2022-06-03:
I looked at the Challenger RP2040 WiFi and as far as I can tell (and test with an LED), the pins are correctly assigned and match the pinout card. SCK, MOSI and MISO are respectively GP22, GP23 and GP24. The problem is that 22 and 23 on the RP2040 are on SPI0 and 24 is on SPI1. Unless I'm missing something, as a result, board.SPI() can not work. That is a board design issue.

If you need to use that group of pins (like for a featherwing), you would have to use bitbangio.
```py
spi = bitbangio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
```
[image]
[image]

NathanY3G, 2022-06-04:
Thanks @Neradoc for the information and potential workaround. It does indeed look like a design issue with this board - pity.

From the CircuitPython perspective should this ideally be cleaned up to try and help others? For example, by disabling / removing the `board.SPI()` constructor. I am happy to try and help with this.

jepler, 2022-06-04:
I wonder how much work it would be to let `board.SPI()` be a bitbangio

PontusO, 2022-06-04:
Hi guys, this is a known problem with our first ever RP2040 design. We have an [errata](https://ilabs.se/challenger-rp2040-wifi-spi-bug/) up on our website with a few workarounds and comments to it. I'd like to contribute here but what do you need us to do in the short run? We've stopped manufacturing this version of the board and new board will no have this problem.

NathanY3G, 2022-06-04:
@jepler's challenge is very appealing :grinning: 
However, if there is a new revision in the works perhaps effort is better spent resolving other issues?

dhalbert, 2022-06-04:
@PontusO If you continue to want to support this version of the board, then you can add a second board definition for the revised board when it arrives. Then I think we should just remove `board.SPI()` here, and leave the rest, assuming the pin names still correspond to the the silkscreen markings

NathanY3G, 2022-06-04:
I have created a pull-request to remove the default SPI bus. I would be grateful if someone could review it for me - thanks! https://github.com/adafruit/circuitpython/pull/6460.

tannewt, 2022-06-06:
Fixed by #6462

closed 2022-06-06 by tannewt