adafruit/circuitpython #6461: Matrix Portal M4 Network object is getting heavier

mpicker90, 2022-06-05:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit Matrix Portal M4 with samd51j19
Board ID:matrixportal_m4
```

### Code/REPL

```python
import gc
import board
import rgbmatrix
import framebufferio
import displayio
from adafruit_display_text import bitmap_label
from adafruit_matrixportal.network import Network

print("starting mem", gc.mem_free())
bit_depth = 2
base_width = 64
base_height = 32
chain_across = 2
tile_down = 1
serpentine = True

width = base_width * chain_across
height = base_height * tile_down

addr_pins = [board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD]
rgb_pins = [
    board.MTX_R1,
    board.MTX_G1,
    board.MTX_B1,
    board.MTX_R2,
    board.MTX_G2,
    board.MTX_B2,
]
clock_pin = board.MTX_CLK
latch_pin = board.MTX_LAT
oe_pin = board.MTX_OE

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=width,
    height=height,
    bit_depth=bit_depth,
    rgb_pins=rgb_pins,
    addr_pins=addr_pins,
    clock_pin=clock_pin,
    latch_pin=latch_pin,
    output_enable_pin=oe_pin,
    tile=tile_down, serpentine=serpentine,
)
display = framebufferio.FramebufferDisplay(matrix)
print("mem after display", gc.mem_free())

group = displayio.Group()
print("mem after display group", gc.mem_free())

#update to font file on system your system
label = bitmap_label.Label(<font file>, anchor_point=(0, 0))
label.color = 0xFF0000
label.text = "hello world"
label.x = 10
label.y = 10
print("mem after label", gc.mem_free())

group.append(label)
print("mem after appending label", gc.mem_free())

display.show(group)
print("mem after display show", gc.mem_free())

network = Network(status_neopixel=board.NEOPIXEL)
print("mem after network creation", gc.mem_free())

response = network.fetch("http://example.com")
print("mem after response", gc.mem_free())
```

### Behavior

This code initializes a display and network object and prints out free memory after each action. When upgrading from CircuitPython 7.2.0 to CircuitPython 7.3.0 i noticed the free memory is reduced by 16,000 at the start of the application. If there is no network connection import, then the free memory is roughly the same. A loss of 16k worth of free memory is quite a bit and causing my board to run out of memory on more complicated applications

### Additional information

Running 7.2.0 without importing network prints
starting mem 112272
mem after display 103824
mem after display group 103744
mem after label 102656
mem after appending label 102656
mem after display show 102656

Running 7.3.0 without importing network prints 
starting mem 112000
mem after display 103552
mem after display group 103472
mem after label 101200
mem after appending label 101200
mem after display show 101200

Running 7.2.0 with a network connection prints 
starting mem 71536
mem after display 63088
mem after display group 63008
mem after label 60816
mem after appending label 60816
mem after display show 60816
mem after network creation 59648
mem after response 58400

Running 7.3.0 with a network connection prints 
starting mem 55712
mem after display 47264
mem after display group 47184
mem after label 44528
mem after appending label 44528
mem after display show 44528
mem after network creation 43456
mem after response 42176

dhalbert, 2022-06-06:
7.2.0 froze four libraries: `adafruit_portalbase`, `adafruit_requests`, `adafruit_esp32spi`, and `neopixel`. We had to turn some features off to make room for all of those.
7.3.0 dropped freezing `adafruit_portalbase`. That is a large library, and is probably the reason you are seeing 16k less RAM, because importing a module (even a .mpy) stores the compiled code in RAM.

We can think about the tradeoffs here.

dhalbert, 2022-07-06:
check for unnecessary imports (used for type-checking only) in `adafruit_requests` and other related libraries

dhalbert, 2022-08-17:
We could probably freeze `adafruit_portalbase` if we turned off `ulab`. This may be a reasonable tradeoff. I found one use of `ulab` in an RGB Matrix project (which was done on the Feather M4, not the MatrixPortal): https://learn.adafruit.com/ocean-epoxy-resin-lightbox-with-rgb-led-matrix-image-scroller/code-explanation, but that could be recoded.

@makermelissa Do you have an opinion about what to refreeze, or other ways to save space?

I have some changes to `adafruit_esp32spi` that will reduce its size, but that PR is in process and needs some debugging and re-testing:
https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/167

Neradoc, 2022-08-17:
`adafruit_portalbase/wifi_esp32s2.py` is small but we could remove it using the same trick as with the Circuit Playground library. Would it make a difference ?

dhalbert, 2022-08-17:
The RAM saved would be roughly the size of the .mpy file, which is only 788 bytes. `adafruit_esp32spi_wsgiserver` is about 2.4k, and I was planning to delete that. But `adafruit_portalbase` is much bigger.

I do see that BLEIO HCI support for the ESP32 is enabled. We could turn that off and that frees a lot of space, more than enough to freeze `adafruit_portalbase`  again. I will ask about this.

PR #6779 (2022-08-23): matrixportal: remove ulab to make room to freeze portalbase
- Fixes #6461, creating more RAM space in MatrixPortal, by restoring `adafruit_portalbase` as a frozen library.

We've removed and then replaced BLE support in MatrixPortal once already:
- #6152
- #6154 

I think it was restored because the MatrixPortal Guide included the standard mirrored AirLift BLE page. However, I see no Learn Guide projects that use it. We can hide or remove that page in the Guide.

@caternuson (author of #6152) Do you know of any specific reason why we need BLE support in MatrixPortal? Note that the AirLift BLE  [... cut]
closed 2022-08-23 by jepler