adafruit/circuitpython #6481: Support for M5Stack devices

scarolan, 2022-06-12:
This feature request is for CircuitPython support on M5Stack devices like the Core, Stick, Atom and Stamp:

https://m5stack.com/

dglaude, 2022-06-12:
Technically, ESP32 are not supported (anymore, since version 3.0?) by CircuitPython. The USB is not native and then it can not support emulating mass storage to expose CIRCUITPY drive. (This is not the case for ESP32 S2 or S3).
Practically, it would be up to the vendor of M5Stack to make the porting and propose a pull request with the support.

tannewt, 2022-06-13:
This depends on: https://github.com/adafruit/circuitpython/issues/1629

Note, we've never had ESP32 support. We did support ESP8266 early on but never the 32.

steveatinfincia, 2022-07-20:
I have most of the M5Stack devices and I've been playing around with the circuitpython source tree to see if it will run on one of them. 

I'm using `ports/espressif/boards/adafruit_feather_esp32_v2` (a regular esp32, which #6613 suggests can at least run to some extent) as a starting point, however I'm not sure what the development workflow for these non-usb devices would be, is `circuitpython-firmware.bin` intended to be written to a particular flash offset using esptool.py?

jepler, 2022-07-20:
Yes, the firmware is to be loaded at 0.  Typical esptool commandline:
```
esptool.py --chip esp32 -p /dev/serial/by-id/usb-1a86_USB_Single_Serial_5479024009-if00 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB 0x0000 build-adafruit_feather_esp32_v2/firmware.bin
```
I don't know the details of the M5Stack. Right now the Feather ESP32 V2 is the only board we have with the ESP32. It has 2MB PSRAM and 8MB flash. I was trying to get the Feather ESP32 V1 (no PSRAM, 4MB flash) working but it just boot loops, so there are more problems to find.

steveatinfincia, 2022-07-21:
The Core 2 works here, took some time to track down the cause of the reset but I just needed to prevent `reset_all_pins()` from resetting pins 7/8, which are psram data pins 0 and 1 on this device.

[image]

steveatinfincia, 2022-07-23:
[image]

Almost everything working on Core 2, including the LCD, speaker, the green LED, vibration motor, internal I2C, and the grove connector.

The LCD on this device is an ili9342c, which is similar to ili9341 but has a few extra initialization commands.

The power management chip on this device is in charge of almost everything, even the green LED, the buttons and the vibration motor. It also supplies power to the speaker, the LCD, and the external ports and has to be initialized before any of those can be used.

I still need to move some of this code down to C so the basic stuff is properly set up in `board_init()`.

tannewt, 2022-07-25:
Nice work @steveatinfincia !

ladyada, 2022-08-01:
neat!

scarolan, 2022-08-05:
Wow this is great.  I have an M5stack Core 2 AWS edition, would love to try CircuitPython and the atecc library with the secure chip:

https://docs.circuitpython.org/projects/atecc/en/stable/

steveatinfincia, 2022-08-05:
I'm getting my core2 branch ready to submit a PR, still fighting with the occasional freezes (I2S, might be a coincidence), but all the major stuff seem to be resolved. Since I started working on this, other contributors have been merging fixes and improvements that made some of the little changes I was making unnecessary, so thank you for that everyone :)

@scarolan They don't provide a schematic for the AWS model so it's hard to tell but I believe the crypto chip is the only difference between it and the Core 2. The M5Stack Arduino libraries don't differentiate between them so that's a good sign.

steveatinfincia, 2022-08-06:
[Here is a development build for the Core 2](https://github.com/steveatinfincia/circuitpython/releases/tag/8.0.0-m5stack-core2-2022.08.05.2) if anyone would like to play around with it.

Also pushed it out on M5Burner 3.0, in the Core 2 firmware area.

It's is a complete 16MB flash image so it takes a little longer to write to the device.

I've been using Thonny 4.0.0b3 to interact with it and it's working pretty well here all things considered.

tannewt, 2022-08-08:
@steveatinfincia Feel free to PR it. We're happy to have follow up PRs for any further fixes.

codegrue, 2022-12-12:
I burned this on the AWS version using M5Burner and it seems to work fine. I don't get a mounted CircuitPython drive though, is that expected behavior?

dhalbert, 2022-12-12:
@codegrue ESP32 and ESP32-C3 chips don't have native general-purpose USB, so a drive cannot be presented. I am not sure which chips are used on which M5Stack devices.

codegrue, 2022-12-13:
Do the ESP based Adafruit "feather" boards support native USB? If not, how would you get circuitpython on those?

anecdata, 2022-12-13:
ESP32-S2 and ESP32-S3 have native USB

ESP32 does not have native USB (so no CIRCUITPY) but:
https://learn.adafruit.com/circuitpython-with-esp32-quick-start

dhalbert, 2022-12-13:
ESP32-S2 and ESP32-S3 boards have native USB support. It is a function of the chip. "ESP32*" is a family of chips, and more confusingly, plain "ESP32" is also the name of the first chip in the family.

This is turning into a support question. Could we continue discussion in our discord, https://adafru.it/discord or the forums, https://forums.adafruit.com/viewforum.php?f=60 ? I will close for now.

dhalbert, 2022-12-13:
Since "M5Stack devices" encompasses a growing family, if there are specific requests, let's make issues for those. Otherwise I wouldn't be sure when we could close the issue.

closed 2022-12-13 by dhalbert

srolf, 2023-06-17:
Even if this is an older issue, is there any chance to get the M5Stack ESP32 GRAY working with new CPy releases?
[https://shop.m5stack.com/products/grey-development-core?variant=16804796006490](https://shop.m5stack.com/products/grey-development-core?variant=16804796006490)
The issue with mine is that it only has got 4MB flash because it's the "old" version. So I guess releases for 16MB like fitting to its "new" version wouldn't work, right?

bill88t, 2023-06-17:
@srolf Well, making a custom board definition and building it locally is always an option.
From there, you could try to make a PR to have it included onto the main repo.

srolf, 2023-06-17:
@bill88t I will give it a try, just being little afraid starting it only after my 5th day with CPy ;-)

bill88t, 2023-06-17:
I suggest you take a read at https://learn.adafruit.com/building-circuitpython/build-circuitpython.
If you need any help I suggest hopping on the Adafruit discord (https://adafru.it/discord) in \#circuitpython-dev.