adafruit/circuitpython #6484: Builtin display invalid after fake deep sleep

Neradoc, 2022-06-13:
There are a few boards that support deep sleep and have a display, but the display becomes unusable after **fake** deep sleep, because the display is not released prior to being re-inited on wake. The [FunHouse does release displays](https://github.com/adafruit/circuitpython/blob/ab346a27fc71f281c9bafb1b684bf00e50f90800/ports/espressif/boards/adafruit_funhouse/board.c#L119) in board_deinit() and doesn't have the issue.

Should that be added to all relevant `board.c` ? Or should a call to release displays be added to the fake sleep code in main.c ? After board_deinit() for example. [Some boards with an e-ink display](https://github.com/adafruit/circuitpython/blob/ab346a27fc71f281c9bafb1b684bf00e50f90800/ports/espressif/boards/adafruit_magtag_2.9_grayscale/board.c#L198-L211) have their own thing where they wait for the display to finish refreshing before releasing.

Example boards that need the fix that I could test:
- adafruit_feather_esp32s2_tft
- hiibot_iots2
- lilygo_ttgo_t8_s2_st7789
- adafruit_macropad_rp2040

jepler, 2022-07-06:
In https://github.com/adafruit/circuitpython/pull/6543 we release displays for all boards.  Please re-test.

Neradoc, 2022-07-06:
Yup, the issue is fixed with the boards I tested.