adafruit/circuitpython #6487: Waveshare RP2040-Zero doesn't mount CIRCUITPY after uf2 install

Zoom67, 2022-06-14:
### CircuitPython version

```python
adafruit-circuitpython-waveshare_rp2040_zero-en_US-7.3.0.uf2
```

### Code/REPL

```python
N/A
```

### Behavior

N/A (see description)

### Description

I'm just trying to get the thing to boot. This is the device:
https://www.waveshare.com/wiki/RP2040-Zero

This is the uf2 I'm trying to load:
https://circuitpython.org/board/waveshare_rp2040_zero/

I can install this uf2 fine, but the Pico and the Waveshare RP2040-Zero have different pin outs and other differences. I'm hoping to use the right uf2 here, but can't get it to work. FWIW, I'm on an M1 Mac.

This will boot:
https://circuitpython.org/board/raspberry_pi_pico/

Steps:
1) Hold BOOTSEL, connect to USB (that is connected to Mac)
2) RPi drive shows up.
3) Drag .uf2 file to the drive
4) RPi drive unmounts but doesn't return as CIRCUITPY

tannewt, 2022-06-15:
This is likely a problem with the flash config. This is the configured flash: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/boards/waveshare_rp2040_zero/mpconfigboard.mk#L9=

closed 2022-06-16 by tannewt

gitterman, 2022-07-16:
Still the issue here on Windows!
I installed adafruit-circuitpython-waveshare_rp2040_zero-en_US-7.3.1.uf2 on my pico zero from https://circuitpython.org/board/waveshare_rp2040_zero/  on my pico zero 
and after that, thonny reports:
_Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter._
I then tried rp2-pico-20220618-v1.19.1.uf2 and this worked ok.
So, I think the issue should be re-opened