adafruit/circuitpython #6490: Please add minimalmodbus

JetForMe, 2022-06-15:
Please add a version of `minimalmodbus` or similar. There was some work done on it, mentioned in the Adafruit forum, but it's not clear where that ended up.

One necessary aspect is control of a GPIO to enable the TX driver on a typical RS-485 driver IC.

Thanks!

sodabrew, 2022-11-14:
Would this work? https://github.com/brainelectronics/micropython-modbus

dhalbert, 2022-11-15:
We share the language core with MicroPython, but the board-specific parts of implementations can be rather different. Some code could probably be adapted. Modbus is not a feature we plan to add any time soon; if a community member wanted it, we'd certainly entertain PR's.

sodabrew, 2022-11-15:
Given that this is an external module, what's the likelihood that it would "just work" vs. needing updates to handle different core APIs?

dhalbert, 2022-11-15:
Is there C native code for Modbus in https://github.com/brainelectronics/micropython-modbus? I see the Python module. Its network code may need to be adapted. But it could be made into a CircuitPython library.

cwworth, 2022-11-29:
Am looking for a CircuitPython port of the Arduino library https://github.com/4-20ma/ModbusMaster library that will enable half-duplex communication using a MAX485 hardware device

toddkarin, 2023-01-29:
Some modbus support would be very helpful! Many high-accuracy devices are only available with modbus.