adafruit/circuitpython #6500: incompatible mpy - adafruit_requests

Craigxyzzy, 2022-06-17:
### CircuitPython version

```python
730 & 8a1  bootloader tiny uf2 094
```

### Code/REPL

```python
from wifi import radio as r
from socketpool import SocketPool as q
from adafruit_requests import Session as p
r.connect("ssid","pass")
print(p(q(r)).get("http://wifitest.adafruit.com/testwifi").text)
```

### Behavior

Traceback (most recent call last):
  File "code.py", line 3, in <module>
MpyError: Incompatible .mpy file.

### Description

220611 adafruit_requests.mpy works fine
220617 adafruit_requests.mpy fails incompatable mpy
220617 adafruit_requests.py  works fine

Tested with 730 & 8a1 on multiple boards; magtag, feather s2 tft, feather s3, um s2, um pro s3

Presume something went wrong with creation of latest mpy file.

domdfcoding, 2022-06-18:
I believe this is due to https://github.com/adafruit/Adafruit_CircuitPython_Bundle/issues/384

Craigxyzzy, 2022-06-18:
.mpy & .py are identical except for version number.
__version__ = "1.12.2"             in .mpy
__version__ = "0.0.0-auto.0"  in .py

Craigxyzzy, 2022-06-18:
Closing as being handled in 384 above.

Thanks for catching that mpy was uncompiled.

closed 2022-06-18 by Craigxyzzy