adafruit/circuitpython #6501: USB fails when REGOUT0 is set below 3.0V

bobxor, 2022-06-18:
I've utilized the nRF52840 Dongle (PCA10059) with nRF52 SDK (Segger) projects to use USB while regulating GPIO voltages at 1.8V.  I'd love to use CircuitPython for an educational target, but I need to regulate the GPIOs to 2.5V.  

I've noticed a curious issue, where the CP USB fails when the REGOUT0 is set below 3.0V.  (It works fine at 3V0 and 3V3 settings)
https://github.com/adafruit/circuitpython/blob/main/ports/nrf/peripherals/nrf/nrf52840/power.c

Frustratingly, the Adafruit bootloader operates at 1.8V but USB filesystem is working correctly.  

I've tested CircuitPython with separate nrf52840 hardware, using external power, and confirmed that the minimum voltage to enable USB is 3.0V at startup.  Curiously, I can dial down the voltage after boot and have USB operation down to ~1.8V.

Is there a workaround to use the PCA10059 and CircuitPython USB below 3V?  Any insight would be greatly appreciated.

dhalbert, 2022-06-20:
> I've noticed a curious issue, where the CP USB fails when the REGOUT0 is set below 3.0V. (It works fine at 3V0 and 3V3 settings)
> https://github.com/adafruit/circuitpython/blob/main/ports/nrf/peripherals/nrf/nrf52840/power.c

This code was added because we encountered cases where all of flash was erased (set to all ones), which caused the I/O voltage to be 1.8v. But if the UICR bits are set specifically to a separate value, and the word is not all ones, this code will not be exercised.

Are you setting REGOUT0.VOUT to 0 or 7? If you set it to 7, then you'll exercise the code above. If you set it to 0, then you should be OK.

Note also "high voltage mode", which is true by default on the PCA10059. See screenshots from the datasheet below.

[image]

[image]

bobxor, 2022-06-20:
I’ve changed this code to set it to 4 (3.0V still works), and also 3 (2.7V CP boots, even runs Boot.py, but fails to enable USB interfaces and halts.)

What I’d like to do is run the system at 2.4V and still have USB - which is something that can be done using Segger and nRF5 SDK.  

I’m starting to wonder if there’s something the TinyUSB implementation is looking for that causes it to be dependent on the main supply regulation.  Nordic documentation clearly shows the USB regulation is separate - enabling USB to work even when main regulation (REGOUT0 is at 1.8V).

dhalbert, 2022-06-20:
What is the wiring to VDD and VDDH? That would determine "high voltage" mode.

dhalbert, 2022-06-20:
I am not so sure this is TinyUSB but you might look at https://github.com/adafruit/Adafruit_nRF52_Bootloader, assuming you are using that bootloader.

bobxor, 2022-06-21:
The PCA10059 I'm using has not been modified, it's operating in the "high voltage" mode.  

Yes, I've using the bootloader.  I've measured the voltages and I can see it's regulating at 1.8V...but the USB works fine.  

After flashing the CircuitPython code, and can verify it changes the regulation to 3.3V.  However, I see the USB fail below 3.0V setting.  

I'll look at how the USB is implemented differently.  It might point to a clue.

dhalbert, 2022-06-21:
Some issues related to this are here:
https://github.com/adafruit/circuitpython/issues/1174
https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/200
https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/48

dhalbert, 2022-10-05:
Closing this for now, but reopen if you have further info.

bobxor, 2022-10-05:
We changed the board design to regulate at 3.3V.  I couldn’t find a workaround with CircuirPython to work below 3.0V.  I’m quite confident it’s related to USB initialization.

chipperdoodles, 2022-12-14:
Just chiming in to say I think I'm having the same issue. I've designed a custom circuitpy board based around a raytac module but it uses the a split 1.8/3.3v rail where the nrf52480 and nvm run off 1.8v (VDD and VDDH are tied so it's in 'normal' mode). My status led's post circuit python uf2 upload seem to be working and suggest it's running, but I get no usb activity so I can't load programs onto it. The adafruit UF2 bootloader works without issue as well while things are in bootloader mode, it's only when the circuit-python application is running that I get no usb detection. I'm more of take an example extend kind of tinkerer but from my trouble shooting @bobxor says about being related to usb bring up might be where the issue? 

I would love to get circuit python running on this board but definitely need usb. I'll keep throwing my head against the wall seeing if i can futz with anything but just chiming in and saying this would be a huge help if it eventually gets addressed! 

-thank you! lmk if you want any more info on the board or usage or testing.

CaptainJackey, 2025-05-23:
> Just chiming in to say I think I'm having the same issue. I've designed a custom circuitpy board based around a raytac module but it uses the a split 1.8/3.3v rail where the nrf52480 and nvm run off [...]

#10365 I'm happy to tell you that this problem has been resolved in this issue.