adafruit/circuitpython #6503: wifi.radio.onnect causes a hard crash on the ESP32-S3-DevKitC-1-N8R2

RetiredWizard, 2022-06-20:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-13-gd0953e8f8 on 2022-06-19; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
```

### Code/REPL

```python
import ssl,wifi,socketpool
import adafruit_requests as requests
from secrets import secrets
wifi.radio.connect(secrets["ssid"], secrets["password"])
```

### Behavior

The ESP32-S3 Devkit will hard crash to safe mode when the wifi.radio.connect call is made. CP 7.3.0 doesn't seem to have the same issue.

RetiredWizard, 2022-06-20:
I dropped back to the previous repository tag 8.0.0-alpha.1:

Adafruit CircuitPython 8.0.0-alpha.1 on 2022-06-19; ESP32-S3-DevKitC-1-N8R2 with ESP32S3

and wifi.radio.connect did not cause this crash.

tannewt, 2022-06-21:
What reason does safe mode give for the problem?

tannewt, 2022-06-22:
Huh, weird that it only safe modes sometimes. The next debugging step would to do a DEBUG=1 build and see what the debug UART has for a backtrace. Want to try that?

RetiredWizard, 2022-06-22:
Sure, I should get a chance to take a look at that this evening.

RetiredWizard, 2022-06-23:
So I've tried a few times but it's not initiating safe mode anymore. The crash still happens at the same point but I don't see anything that looks like a backtrace in the debug output.

> I (39930) pp: pp rom version: e7ae62f
I (39930) net80211: net80211 rom version: e7ae62f
I (39940) wifi:wifi driver task: 3fce4ee8, prio:23, stack:3584, core=0
I (44866) system_api: Base MAC address is not set
I (44866) system_api: read default base MAC address from EFUSE
I (44876) wifi:wifi firmware version: e2cf259
I (44876) wifi:wifi certification version: v7.0
I (44886) wifi:config NVS flash: disabled
I (44886) wifi:config nano formating: disabled
I (44886) wifi:Init data frame dynamic rx buffer num: 8
I (44896) wifi:Init management frame dynamic rx buffer num: 8
I (44906) wifi:Init management short buffer num: 32
I (44906) wifi:Init dynamic tx buffer num: 16
I (44906) wifi:Init tx cache buffer num: 16
I (44916) wifi:Init static tx FG buffer num: 2
I (44916) wifi:Init static rx buffer size: 1600
I (44926) wifi:Init static rx buffer num: 4
I (44926) wifi:Init dynamic rx buffer num: 8
I (44926) wifi_init: rx ba win: 6
I (44936) wifi_init: tcpip mbox: 32
I (44936) wifi_init: udp mbox: 6
I (44946) wifi_init: tcp mbox: 6
I (44946) wifi_init: tcp tx win: 2880
I (44946) wifi_init: tcp rx win: 2880
I (44956) wifi_init: tcp mss: 1440
I (44956) wifi_init: WiFi IRAM OP enabled
I (44966) wifi_init: WiFi RX IRAM OP enabled
I (44966) phy_init: phy_version 501,79e7e9b,Apr 19 2022,11:10:08
E (44976) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
W (44986) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
I (45076) wifi:mode : sta (7c:df:a1:e0:71:34)
I (45076) wifi:enable tsf
W (45076) wifi: sta start
I (72146) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (72846) wifi:state: init -> auth (b0)
I (72846) wifi:state: auth -> assoc (0)
I (72856) wifi:state: assoc -> run (10)
W (72876) wifi:<ba-add>idx:0 (ifx:0, 4a:5d:36:d1:30:e1), tid:0, ssn:0, winSize:64
I (72886) wifi:connected with twilightzone-guest, aid = 9, channel 1, BW20, bssid = 4a:5d:36:d1:30:e1
I (72886) wifi:security: WPA2-PSK, phy: bgn, rssi: -76
I (72896) wifi:pm start, type: 1

> I (72896) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
W (72906) wifi: connected                             <<<<<<<< CRASHES HERE >>>>>>>>>>>
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40374200
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0108,len:0x1650
load:0x403b6000,len:0xbb8
load:0x403ba000,len:0x2f78
entry 0x403b6254
I (31) boot: ESP-IDF 944c01eef 2nd stage bootloader
I (32) boot: compile time 23:35:34
I (32) boot: chip revision: 0
I (34) boot.esp32s3: Boot SPI Speed : 40MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
W (48) boot.esp32s3: PRO CPU has been reset by WDT.
W (53) boot.esp32s3: APP CPU has been reset by WDT.
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (83) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (90) boot:  2 ota_0            OTA app          00 10 00010000 00200000
[... 32 lines cut ...]
I (738) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (745) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (751) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (758) spi_flash: detected chip: gd
I (761) spi_flash: flash io: dio
I (766) sleep: Configure to isolate all GPIO pins in sleep state
I (772) sleep: Enable automatic switching of GPIO sleep configuration
I (780) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0

One of the times I crashed the board there was on additional log line between the "wifi:set rx beacon" and "wifi:connected" lines:

> I (161471) wifi:BcnInt:102400, DTIM:1

I tried a dozen or two times varying what I could, storage.remount, pasting wifi script or importing from flash, etc trying to get a safe mode reboot without any luck. I did get a failed to connect reason code 4 and no board crash on one attempt.

> W (32479) wifi: sta start
I (34699) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (35399) wifi:state: init -> auth (b0)
I (35409) wifi:state: auth -> assoc (0)
I (36409) wifi:state: assoc -> init (400)
I (36409) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
W (36409) wifi: disconnected
W (36409) wifi: reason 4 0x04

anecdata, 2022-06-23:
Reason [4](https://github.com/jepler/esp-idf/blob/d06744f5efc382c61cbad8758107cec308feef09/components/esp_wifi/include/esp_wifi_types.h) is Association Expired, which is probably routine. It should have tried to reconnect several times after that.

tannewt, 2022-06-23:
> rst:0x8 (TG1WDT_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)

Yup, this is what I've seen too. I'll poke at this after the web workflow stuff is out for PR.

RetiredWizard, 2022-06-24:
Playing around a bit more tonight and I did get one safe mode reboot which yielded the following backtrace:

> I (277665) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
W (277675) wifi: connected

> Backtrace:0x4201E385:0x3FCF4F800x4201EEE3:0x3FCF4FA0 0x4201E129:0x3FCF4FC0 0x4201E233:0x3FCF4FE0 0x40380BBA:0x3FCF5000 0x42028BE1:0x3FCF5020 0x4201E5CA:0x3FCF5040 0x4201F36F:0x3FCF5060 0x4202A241:0x3FCF5080 0x42027475:0x3FCF50B0 0x4201210E:0x3FCF5120 0x4200DF0E:0x3FCF5150 0x4200DFFD:0x3FCF5170 0x4201B91A:0x3FCF5190 0x420121A2:0x3FCF5230 0x4200DF0E:0x3FCF5260 0x4200DF32:0x3FCF5280 0x42032052:0x3FCF52A0 0x4203237D:0x3FCF5330 0x4201DDC4:0x3FCF53D0 0x4201E19A:0x3FCF53F0 0x4201E4EB:0x3FCF5420 0x420B6D77:0x3FCF5440

> I (2ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037bd54
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0108,len:0x1650
load:0x403b6000,len:0xbb8
load:0x403ba000,len:0x2f78
entry 0x403b6254

I then threw a couple mp_printf statements before and after the connection loop in common_hal_wifi_radio_connect and the connections started working, however pressing ctrl-D and attempting the connection script a second time, resulted in the same crash.

I suppose this was obvious anyway but when the board does crash the message prior to the connection loop prints but I never get the message following the loop.

    mp_printf(&mp_plat_print, "Starting Connect Loop\n");
    do {
        RUN_BACKGROUND_TASKS;
        bits = xEventGroupWaitBits(self->event_group_handle,
            WIFI_CONNECTED_BIT | WIFI_DISCONNECTED_BIT,
            pdTRUE,
            pdTRUE,
            0);
        // Don't retry anymore if we're over our time budget.
        if (self->retries_left > 0 && common_hal_time_monotonic_ms() > end_time) {
            self->retries_left = 0;
        }
    } while ((bits & (WIFI_CONNECTED_BIT | WIFI_DISCONNECTED_BIT)) == 0 && !mp_hal_is_interrupted());
    mp_printf(&mp_plat_print, "Out of Connect Loop\n");

RetiredWizard, 2022-06-24:
Rather than pressing Ctrl-D after the original connection script and re-running the script to cause the crash, I was able to cause the crash by running the following script:

     >>> import ssl,wifi,socketpool
     >>> import adafruit_requests as requests
     >>> from secrets import secrets
     >>> wifi.radio.connect(secrets["ssid"], secrets["password"])
     Starting Connect Loop
     Out of Connect Loop
     >>> wifi.radio.enabled=False
     >>> wifi.radio.enabled=True
     >>> wifi.radio.connect(secrets["ssid"], secrets["password"])
     Starting Connect Loop

tannewt, 2022-06-24:
Could you use `ports/espressif/tools/decode_backtrace.py <board name>` to decode the backtrace? (Copy and paste the `Backtrace: ` line into it.) The addresses will vary with your build. Thanks!

RetiredWizard, 2022-06-24:
Unfortunately I didn't realize how the backtrace decode worked and I don't think the build I have is the same. Although it's possible the only change I had made is adding the debug print statements in Radio.c. If I'm reading the backtrace decode correct the statement being pointed to is:

      xTaskNotifyWait(0x01, 0x01, NULL, portMAX_DELAY); 

Which actually seems plausible. The decoded backtrace is:

> /home/pi/circuitpython/ports/espressif/supervisor/port.c:292
/home/pi/circuitpython/ports/espressif/../../main.c:699
/home/pi/circuitpython/ports/espressif/../../main.c:1000
/home/pi/circuitpython/ports/espressif/build-espressif_esp32s3_devkitc_1_n8r2/esp-idf/../../esp-idf/components/esp_hw_support/include/soc/spinlock.h:158
/home/pi/circuitpython/ports/espressif/common-hal/microcontroller/__init__.c:79
/home/pi/circuitpython/ports/espressif/../../supervisor/shared/background_callback.c:91
/home/pi/circuitpython/ports/espressif/../../supervisor/shared/tick.c:118
/home/pi/circuitpython/ports/espressif/common-hal/wifi/Radio.c:306
/home/pi/circuitpython/ports/espressif/../../shared-bindings/wifi/Radio.c:355
/home/pi/circuitpython/ports/espressif/../../py/objfun.c:136
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:665
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:680
/home/pi/circuitpython/ports/espressif/../../py/vm.c:992
/home/pi/circuitpython/ports/espressif/../../py/objfun.c:296 (discriminator 4)
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:665
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:638
/home/pi/circuitpython/ports/espressif/../../shared/runtime/pyexec.c:128
/home/pi/circuitpython/ports/espressif/../../shared/runtime/pyexec.c:720
/home/pi/circuitpython/ports/espressif/../../main.c:818
/home/pi/circuitpython/ports/espressif/../../main.c:917
/home/pi/circuitpython/ports/espressif/supervisor/port.c:383

RetiredWizard, 2022-06-24:
I went ahead and rebuilt the firmware without the debug print statements and the resulting decode backtrace is:

> /home/pi/circuitpython/ports/espressif/supervisor/port.c:296
/home/pi/circuitpython/ports/espressif/../../main.c:840
??:?
/home/pi/circuitpython/ports/espressif/build-espressif_esp32s3_devkitc_1_n8r2/esp-idf/../../esp-idf/components/esp_hw_support/include/soc/spinlock.h:158
/home/pi/circuitpython/ports/espressif/common-hal/microcontroller/__init__.c:83
/home/pi/circuitpython/ports/espressif/../../supervisor/shared/background_callback.c:96
/home/pi/circuitpython/ports/espressif/../../supervisor/shared/tick.c:123
/home/pi/circuitpython/ports/espressif/common-hal/wifi/Radio.c:311
/home/pi/circuitpython/ports/espressif/../../shared-bindings/wifi/Radio.c:356
/home/pi/circuitpython/ports/espressif/../../py/objfun.c:136
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:665
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:681
/home/pi/circuitpython/ports/espressif/../../py/vm.c:1021
/home/pi/circuitpython/ports/espressif/../../py/objfun.c:298 (discriminator 4)
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:665
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:639
/home/pi/circuitpython/ports/espressif/../../shared/runtime/pyexec.c:149
/home/pi/circuitpython/ports/espressif/../../shared/runtime/pyexec.c:739
/home/pi/circuitpython/ports/espressif/../../main.c:820
/home/pi/circuitpython/ports/espressif/../../main.c:919
/home/pi/circuitpython/ports/espressif/supervisor/port.c:392
/home/pi/circuitpython/ports/espressif/build-espressif_esp32s3_devkitc_1_n8r2/esp-idf/../../esp-idf/components/freertos/port/port_common.c:131

I'm not sure either is correct. I guess I'll try a bit more to cause a new safe mode crash and grab a backtrace decode from the build I know is in sync.

RetiredWizard, 2022-06-25:
I rebuilt a fresh DEBUG=1 image and flashed it and after a lot of attempts finally managed to create a crash into safe mode and caught the back trace. The decode back trace is:

> /home/pi/circuitpython/ports/espressif/supervisor/port.c:294
/home/pi/circuitpython/ports/espressif/../../main.c:994
/home/pi/circuitpython/ports/espressif/../../main.c:1001
/home/pi/circuitpython/ports/espressif/build-espressif_esp32s3_devkitc_1_n8r2/esp-idf/../../esp-idf/components/esp_hw_support/include/soc/spinlock.h:158
/home/pi/circuitpython/ports/espressif/common-hal/microcontroller/__init__.c:82
/home/pi/circuitpython/ports/espressif/../../supervisor/shared/background_callback.c:94
/home/pi/circuitpython/ports/espressif/../../supervisor/shared/tick.c:122
/home/pi/circuitpython/ports/espressif/common-hal/wifi/Radio.c:310
/home/pi/circuitpython/ports/espressif/../../shared-bindings/wifi/Radio.c:355
/home/pi/circuitpython/ports/espressif/../../py/objfun.c:136
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:665
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:680
/home/pi/circuitpython/ports/espressif/../../py/vm.c:1021
/home/pi/circuitpython/ports/espressif/../../py/objfun.c:297 (discriminator 4)
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:665
/home/pi/circuitpython/ports/espressif/../../py/runtime.c:638
/home/pi/circuitpython/ports/espressif/../../shared/runtime/pyexec.c:146
/home/pi/circuitpython/ports/espressif/../../shared/runtime/pyexec.c:745
/home/pi/circuitpython/ports/espressif/../../main.c:227
/home/pi/circuitpython/ports/espressif/../../main.c:403
/home/pi/circuitpython/ports/espressif/../../main.c:927
/home/pi/circuitpython/ports/espressif/supervisor/port.c:391
/home/pi/circuitpython/ports/espressif/build-espressif_esp32s3_devkitc_1_n8r2/esp-idf/../../esp-idf/components/freertos/port/port_common.c:129

tannewt, 2022-06-27:
app_main is the top of the stack. (Am I printing it out backwards?)

What is at `/home/pi/circuitpython/ports/espressif/supervisor/port.c:294`?

RetiredWizard, 2022-06-27:
I really doubt you're reading it backwards :D, someone else must have....

port.c:294:

        292   void reset_cpu(void) {
        293       #ifndef CONFIG_IDF_TARGET_ESP32C3
        294       esp_backtrace_print(100);
        295       #endif
        296      esp_restart();
        297   }

RetiredWizard, 2022-06-28:
oh wait, <light dawns> you were asking if the decoded output was backwards.... well I think normally I read the last line of a trace to find the last executed statements. It appears that the output from decode_backtrace is the other way around. That's not really a problem as long as you know which way to read it.

The port.c:294 code looks to me like it's just dumping the backtrace, the referenced code segments prior to that one are:

===============================================
main.c:994 1001

    #ifndef NDEBUG
    static void NORETURN __fatal_error(const char *msg) {
        reset_into_safe_mode(MICROPY_FATAL_ERROR);          <----994
        while (true) {
        }
    }

    void MP_WEAK __assert_func(const char *file, int line, const char *func, const char *expr) {
        mp_printf(&mp_plat_print, "Assertion '%s' failed, at file %s:%d\n", expr, file, line);
        __fatal_error("Assertion failed");    <------------- 1001
    }
    #endif

================================================
ports/espressif/esp-idf/components/esp_hw_support/include/soc/spinlock.h:158

    #if !CONFIG_FREERTOS_UNICORE && !BOOTLOADER_BUILD
        uint32_t irq_status;
        uint32_t core_id;

        assert(lock);
        irq_status = XTOS_SET_INTLEVEL(XCHAL_EXCM_LEVEL);

        RSR(PRID, core_id);
        assert(core_id == lock->owner); // This is a mutex we didn't lock, or it's corrupt  <---- 158
        lock->count--;

        if(!lock->count) {
            lock->owner = SPINLOCK_FREE;
        } else {
            assert(lock->count < 0x100); // Indicates memory corruption
        }

        XTOS_RESTORE_INTLEVEL(irq_status);
    #endif

The comment on line 158 of spinlock suggests that something may have already gone wrong at this point, but it looks to me like there should have been an Assertion failed messaged printed which I'm pretty sure I never saw. 

The traced code prior to these also has some concerning comments:

=======================================================
/home/pi/circuitpython/ports/espressif/common-hal/microcontroller/init.c:82

    void common_hal_mcu_disable_interrupts(void) {
        if (nesting_count == 0) {
            portENTER_CRITICAL(&cp_mutex);
        }
        nesting_count++;
    }

    void common_hal_mcu_enable_interrupts(void) {
        if (nesting_count == 0) {
            // Maybe log here because it's very bad.
        }
        nesting_count--;
        if (nesting_count > 0) {
            return;
        }
        portEXIT_CRITICAL(&cp_mutex);   <------ 82
    }

nesting_count appears to have reached zero, or it could have gone negative?

RetiredWizard, 2022-06-28:
I was thinking that the reason I wasn't seeing the assertion message was that the USB interface was crashing so I rebuilt with debug uart enable and connected a terminal via RX/TX. 

When connecting via an RX/TX terminal, the board doesn't crash. I should have made this connection earlier but that reminded me of a Micropython wifi crash issue https://github.com/micropython/micropython/issues/8635. It looks like the Micropython issue was resolved by moving to a newer ESP IDF.

RetiredWizard, 2022-06-28:
After some more testing.... it turns out that when connected via an RX/TX connection rather than over UART, the wifi connect usually didn't crash the board, however if I ran the wifi connect repeatedly, it did eventually crash the board.

The comment in /home/pi/circuitpython/ports/espressif/common-hal/microcontroller/__init__.c concerned me so I threw a return in the if block:

    if (nesting_count == 0) {
        // Maybe log here because it's very bad.
        return;
    }

This didn't seem to have any impact when connecting via USB, however I haven't been able to crash the board even after repeatedly running the wifi connect code with this modification. I suspect it's not going to matter as hopefully an update to the IDF will resolve the issue completely anyway.

tannewt, 2022-06-28:
It does look like we're on 4.4 with a cherry picked fix for I2C. I'll look into updating the IDF today.

tannewt, 2022-06-28:
I've got an IDF update PR here: https://github.com/adafruit/circuitpython/pull/6531

RetiredWizard, 2022-06-29:
I ran the following steps from Linux:

1. git clone https://github.com/adafruit/circuitpython.git
2. git fetch origin pull/6531/head:TESTRW
3. git checkout TESTRW
4. make fetch-submodules

And then built as I usually do. During the fetech-submodules phase it looked to me like the IDF updated to 4.4.1 but I don't know how to verify if I'm on the correct IDF version and unfortunatly, I'm still getting a crash on the wifi connect when connected via USB.

RetiredWizard, 2022-06-29:
I'm tied up for a few hours how, but I have been trying various tweaks without any success but as one last thing I decided to do a DEBUG=1 build to see if the backtrace is much different and in the debug build the connect command is not crashing.

I've got most of the build disabled so that the DEBUG build would fit. perhaps the current conflict is coming from one of the modules I've disabled. Unless you come up with another breakthrough, this evening maybe I'll try adding back in a module at a time and see if I can isolate the conflict.

tannewt, 2022-06-29:
I figured it out! It's a multicore issue. With the auto-wifi stuff I added a call to `supervisor_workflow_request_background()` in `event_handler` in `common-hal/wifi/__init__.c`. This handler runs on the PRO CPU and CP is on the APP CPU. So, we need to do the schedule on the APP CPU. (We could do proper locking but that's more work.) Fix coming shortly. Thanks for helping investigate this!

closed 2022-06-29 by dhalbert

RetiredWizard, 2022-06-29:
Nice work Scott!! I just downloaded and built after Dan's merges and it seems to be working.

PR #6533 (2022-06-29): Fix auto-wifi created crash
The wifi event_handler runs on the other core so we need to be
careful when calling into CP APIs.

Fixes #6503
commit 8dc82652 (2022-06-29): Fix auto-wifi created crash

The wifi event_handler runs on the other core so we need to be
careful when calling into CP APIs.

Fixes #6503