adafruit/circuitpython #6504: Getting Message "loc" problem

SeekerJBP, 2022-06-20:
Worked just a few days ago, but when using update script, I get this message:

There was a problem: local variable 'loc' referenced before assignment

It has always worked before, and I ty to keep everything up to date... so I am not sure what has changed.

This is the message in full:

"C:\Users\xyz\Desktop>circup update --all
Found device at E:\, running CircuitPython 7.3.0.
There was a problem: local variable 'loc' referenced before assignment"

Any ideas?

SeekerJBP, 2022-06-20:
Forgot to add the script it reads:

pip install --upgrade circup
circup update --all

Like I said, worked 2 or 3 days ago, so wondering what changed.

Neradoc, 2022-06-20:
Hi, thanks for reporting. This is a bundle and circup issue. See https://github.com/adafruit/circup/issues/146
The bundle issue [is fixed](https://github.com/adafruit/circuitpython-build-tools/pull/88), and the Bundles have been re-released.

Unfortunately you'll have to manually delete the libraries, and reinstall them.
You can automatically install the libraries used by `code.py` like this:
```
circup install --auto
```
As well as by any other file on the board:
```
circup install --auto --auto-file other_file.py
```

tannewt, 2022-06-21:
This is fixed thanks to @Neradoc

closed 2022-06-21 by tannewt