adafruit/circuitpython #6505: Issues with bringing up the new Pimoroni Tufty LCD board

ZodiusInfuser, 2022-06-20:
### CircuitPython version

```python
boot_out.txt reports --> Adafruit CircuitPython 6.2.0-beta.2-1233-g38015b6ca-dirty on 2022-06-16; Pimoroni Tufty 2040 with rp2040
Board ID:pimoroni_tufty2040

Note, I'm not convinced that the CircuitPython is 6.2.0, as this repo is only 37 commits behind adafruit:main

Repo for custom build: https://github.com/ZodiusInfuser/circuitpython/tree/tufty
```

### Code/REPL

```python
import time
import board
import displayio
import terminalio
import vectorio
from adafruit_display_text import label

print("Starting...")
#time.sleep(5)

display = board.DISPLAY
#display.brightness = 0.01

# Set text, font, and color
title = "HELLO WORLD"
subtitle = "From CircuitPython"
font = terminalio.FONT
color = 0xFF0000

# Set the palette for the background color
palette = displayio.Palette(1)
palette[0] = 0xFFFFFF

# Add a background rectangle
rectangle = vectorio.Rectangle(pixel_shader=palette, width=display.width, height=display.height, x=0, y=0)

# Create the title and subtitle labels
title_label = label.Label(font, text=title, color=color, scale=4)
subtitle_label = label.Label(font, text=subtitle, color=color, scale=2)

# Set the label locations
title_label.x = 30
title_label.y = 70

subtitle_label.x = 50
subtitle_label.y = 160

# Create the display group and append objects to it
group = displayio.Group()
group.append(rectangle)
group.append(title_label)
group.append(subtitle_label)

# Show the group and refresh the screen to see the result
display.show(group)
#display.refresh()

time.sleep(10)
```

### Behavior

See video clip: https://twitter.com/ZodiusInfuser/status/1537201231608807430?s=20&t=bn2rUOjI8ZFX1rcEbIGn7Q

### Description

Hi. I was directed here by @tannewt on Twitter, as I have been having this odd issue with the CircuitPython bring-up for the new Pimoroni Tufty 2040, an LCD name badge that uses parallel bus.

Basically, after much head scratching and probing I was able to port our display initialisation from C++/MP over to a CPy DisplayIO object, and am greeted with the onscreen REPL. However, when trying to draw things on the display, if I happen to draw a mostly or full white rectangle over 70% of the screen I end up with what appears to be image smearing. I have no clue why this is. If I draw a more primary colour it is fine. Has this been encountered before?

I suspect this is something to do with my setup rather than CircuitPython itself, but much of the underlying binding functions are a black box to me, on top of the general black box of LCDs, so I am a loss of how to diagnose this. I have confirmed by logic analyser that the init sequence is sending the correct bytes.

One thing I did notice is that CPy draws a row at a time, whereas our C++/MP code draws the full screen. I wonder if there is a way to make CPy draw the full screen at once too to speed it up to the display's native 60Hz?

### Additional information

Product -> https://shop.pimoroni.com/products/tufty-2040
Board file -> https://github.com/ZodiusInfuser/circuitpython/blob/tufty/ports/raspberrypi/boards/pimoroni_tufty2040/board.c

ZodiusInfuser, 2022-06-20:
I've done some more scoping tonight, and best I can tell, the data being sent to the display is clean. Correct number of bytes sent, correct pattern. The only difference between the one that works and the one that doesn't is one receives `0xffe0` for each background byte (yellow in 16bit), and the problem case receives `0xffff`.

This puts it back to being something wrong with my initialisation, but as I said before, it matches what we do in C++ and MP. Could it be that the initialisation is slightly wrong, but it only manifests itself here 🤔

ZodiusInfuser, 2022-06-20:
Since the init looked correct, as a test I tried reducing the parallel bus speed from 10MHz down to 1.25MHz, and that seems to remove the ghosting, suggesting there's some timing issue trying to drive it faster (not sure if on the display side or mcu side).

I would consider this fixed, were it not for the random white pixels it introduces into the last column of the on-screen REPL, and the fact it's well below the speed needed to achieve 60Hz.

tannewt, 2022-06-21:
Thanks for filing an issue!

> One thing I did notice is that CPy draws a row at a time, whereas our C++/MP code draws the full screen. I wonder if there is a way to make CPy draw the full screen at once too to speed it up to the display's native 60Hz?

CP does one row at a time to reduce the memory needed. It actually track dirty rectangles and only sends them. Unfortunately, the terminal tends to dirty most of the screen. The code will split large dirty rectangles into smaller ones so that we need less RAM while computing pixel values.

I don't think this is a power issue because the display is backlit.

It feels like boundary condition issue to me. Have you confirmed that CS is held low long enough as data is written? Slowing down the PIO shouldn't help that though...

Another thing to check would be the pin drive strength. Maybe that would cause slower speeds to work better. Do you have the ability to get an analog trace?

I definitely haven't seen this exact issue so I'm just brainstorming.

bablokb, 2022-10-07:
I have built this for 8.0.0-beta1 and I can confirm the ghosting with the high bus speed. Ghosting disappears when reducing the speed. But there are still artifacts. The `display_shapes_circle_animation.py` example is a nice test. While moving across the  screen, the circle sometimes leaves a trail of artifacts and sometimes not. This is independent of background color. The artifacts seem to be black on a white background and purple on a black background.

BTW: the artifacts are reproducible on every run. So I don't think it is a (random) timing problem.

ajnbrs, 2022-11-08:
FYI I get this issue using Pimoroni's picographics library in Micropython as well, I got around it by changing my white to very slightly off white, somewhere around rgb(240, 240, 240) works fine but going much closer to pure white and I have the issue

bablokb, 2022-11-09:
This confirms what I think: it is a firmware issue. The CircuitPython code looks correct and I tested slight variations in speed and timings, but that does not change anything fundamental: either I have artifacts (slower bus speed) or ghosting (high bus speed). In the latter case, ghosting disappears when keeping away from white.

ajnbrs, 2022-11-09:
Initially thought it was something in my code as I dropped a fairly complex script over from another display.. I have now added an issue over at Pimoroni as well with some rudimentary examples..

ZodiusInfuser, 2022-11-10:
@ajnbrs Any chance you can share your Micropython code that produced this issue?