adafruit/circuitpython #6510: SPI dropping least significant bit on ESP32-S2 but works on PI Pico

CTGControls, 2022-06-21:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0 on 2022-05-23; Adafruit Metro ESP32S2 with ESP32S2
Board ID:adafruit_metro_esp32s2
```

### Code/REPL

```python
Not working esp32-2 repo
https://github.com/CTGControls/CTG_DPS_Python_Ad4116_Adafruit_MetroExpressEsp32S2

Working PI Pico repo:
https://github.com/CTGControls/CTG_DPS_Python_Ad4116_RaspberryPiPico
```

### Behavior

## I get:
get_register: got [ 0x80 0x00 0x54 ] from reg [ 0x30 ] using command [ 0x70  ] 

## but should get
get_register: got [ 0x80 0x00 0x55 ] from reg [ 0x30 ] using command [ 0x70  ] 

### Description

Only happens on the Metro ESP32-S2 but not on a PI Pico.

dhalbert, 2022-06-21:
The AD4116 datasheet says:

> DIGITAL COMMUNICATION
> The AD4116 has a 3-wire or 4-wire SPI that is compatible with
> QSPI™, MICROWIRE®, and DSPs. The interface operates in SPI
> Mode 3 and can be operated with CS tied low. In SPI Mode 3,
> SCLK idles high, the falling edge of SCLK is the drive edge, and
> the rising edge of SCLK is the sample edge. Data is clocked out
> on the falling and drive edge and data is clocked in on the rising
> and sample edge.

Given that it says it operates in mode 3 (opposite phase and polarity from standard SPI), and with CS tied low, not using `adafruit_busdevice` makes sense. Just just `busio.SPI`, and use `.try_lock()` to grab the bus. The opposite phase and polarity  can cause difficulties because the "normal" states are different from the initial resting states of the pins. It may be a phase/polarity issue in the Espressif implementation. I brought this up during an internal meeting today, and was told that we have had trouble running SPI displays with mode 3 and CS pin tied down, so we in general don't support that kind of operation. Some cheap displays don't expose the CS pin.

CTGControls, 2022-06-21:
This is rather interesting because many of the high end Analog Device chips do use mode 3 only.  Such as:
- AD411x (AD4111, AD4112, AD4113, AD4114, AD4115, AD4116)
- AD717x (AD4172, AD7173, AD7175, AD7175, AD7176, AD7177)

Others use both Mode 0 and Mode 3. Such as:
- AD3552R (give different results if you use one mode over the other)

It looks like AD is using Mode 3 for most of there newest chips as well. I normally always use mode 3 if available as I have seen better results. 

As this is only my second week using Python I am trying to get up to speed.

CTGControls, 2022-09-16:
If you send me an address I will send you an EVAL-AD4116ASDZ (https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/EVAL-AD4116.html) 

Christopher Gay
Chris@CTGControls.com

dhalbert, 2022-09-16:
Thank you for the offer. I will test with an ADXL343 (https://www.adafruit.com/product/4097), which is an Analog Devices accelerometer that uses mode 3 SPI first, and see if I can reproduce. I would think I can.

dhalbert, 2022-11-18:
I did some testing with an ADXL343. I discovered a minor problem in the mode 3 setup on the RP2040 (the clock starts out low), but I have a fix for that and will be PR'ing it.

On the ESP32-S2, I don't have a good analog of your example. I am getting data from the ADXL343 which looks OK, including having bytes with the LSB = 1, but it is not the last byte that is read (which happens to be zero).

There are various Espressif bug reports about mode 3. The most notable one is https://github.com/espressif/esp-idf/issues/7825, but I am not sure whether it is related.

Some questions:
1. Are you seeing this problem only in the last byte read of a multibyte data read?
2. Do you see this problem when you read a single byte whose LSB should be 1?
3. What is in register 0x30 (or really 0x32, I guess)? Is that an ID, some setting, or a data value? I would try to do something similar with the ADXL343.
4. Do you see this problem if you lower the clock rate a lot?
5. Have you tried duplicating on ESP32-S2 Arduino?
6. Does the device work fine in I2C mode? Is that a workaround for you or is it too slow?
7. Do you have waveform traces from a Saleae or similar that show the problem? It would be good to compare the traces on the RP2040 and ESP32-S2.

Thanks.