adafruit/circuitpython #6525: Add support for LilyGo TTGO T-Display

scarolan, 2022-06-24:
A great little board, comes with two buttons, a LiPo connector, and a 240x135 color TFT:
https://github.com/Xinyuan-LilyGO/TTGO-T-Display

It looks like the TTGO T8 ESP32-S2 is already supported. 
https://circuitpython.org/board/lilygo_ttgo_t8_s2_st7789

todbot, 2022-06-24:
This looks like it's ESP32-based, not an ESP32-C3 or -S2 or -S3. 
CircuitPython has not targeted normal ESP32 boards

tannewt, 2022-06-27:
We've started working on ESP32 support but won't add this board ourselves because us Adafruit folks don't have one. So, please add it once we have ESP32 support merged in. Thanks!

scarolan, 2022-08-29:
Wow!  ESP32 support is rad.  Can I donate a board for you to test with?  These are such fun little boards, the screen is so sharp!

ladyada, 2022-08-29:
hiya, we dont take board donations - we're already drowning in PCBs :) instead you can follow our guide
https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

or perhaps collaborate with someone else who has this board :)
https://github.com/adafruit/circuitpython/pulls?q=is%3Apr+is%3Aclosed+ttgo

scarolan, 2023-02-10:
Thank you @ladyada - I'm going to close this out.  There are plenty of other options now in your 366 (!) supported boards!

https://circuitpython.org/downloads

closed 2023-02-10 by scarolan