adafruit/circuitpython #6527: add ability to get "USB-MIDI Event Packets" for USB HOST implementation

fadh3r, 2022-06-28:
I can't get midi data through current endpoint of usb host implementation because midi data is sending with "USB-MIDI Event Packets".
Am I right or I am watching in wrong direction?

fadh3r, 2022-06-28:
this is my code:
`import usb
import usb_host
import board
import digitalio
import microcontroller
import time
import array

#TEST LED
p7 = digitalio.DigitalInOut(board.LED)
p7.direction = digitalio.Direction.OUTPUT
p7.value=True

#INIT USB_HOST WITH POSITIVE AND NEGATIVE PINS
usb_dp = microcontroller.pin.USB_OTG2_DP
usb_dm = microcontroller.pin.USB_OTG2_DN
usb_port = usb_host.Port(usb_dp, usb_dm)

#GIVE POWER TO USB DEVICE
usb_power_on = digitalio.DigitalInOut(board.USB_HOST_POWER)
usb_power_on.direction = digitalio.Direction.OUTPUT
usb_power_on.value=True

#FIND VMETER
vmeter = usb.core.find(idVendor=0x16d0, idProduct=0x0668)
print(vmeter.manufacturer, vmeter.product)

if vmeter.is_kernel_driver_active(0):
    vmeter.detach_kernel_driver(0)

buf = array.array('b', [0] * 64)

while True:
    try:
        count = vmeter.read(0x82, buf)
        # bEndpointAddress 0x81 IN Endpoint 2
        # bEndpointAddress 0x01 OUT Endpoint 1
    except usb.core.USBTimeoutError:
        continue

    print(buf)
    time.sleep(.01)`

I'm getting device manufacturer and product right, but my buf is always filled with zeros...

fadh3r, 2022-06-28:
and here is info about device:
'''
Endpoint Descriptor (Audio/MIDI 1.0):
------------------------------
0x09	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x01	bInterval         
0x00	bRefresh
0x00	bSynchAddress
'''

tannewt, 2022-06-28:
read isn't implemented yet. :-) I mentioned it here: https://github.com/adafruit/circuitpython/pull/6125

I should have had it raise an error. Sorry!

I thought I'd get back to it quickly but didn't. I switched off of it while waiting for a TinyUSB API to call.

Would you like to try and add it? It shouldn't be too hard now that TinyUSB has an API for it. (I haven't looked at specifics though.)

fadh3r, 2022-06-29:
Thanks for answer, Scott!
Sure, I want to try to add it. But can you give me brief insight into right direction, please?
Do I need build new circuitpython for my teensy, or I can just add TinyUSB library to my project and write some glue?

tannewt, 2022-06-29:
> Sure, I want to try to add it. But can you give me brief insight into right direction, please?

Yup! Happy to. Join us on discord if you haven't yet too. There is a #circuitpython-dev channel where we can help you get over any bumps. https://adafru.it/discord

> Do I need build new circuitpython for my teensy, or I can just add TinyUSB library to my project and write some glue?
Yes, you'll need to [build your own CircuitPython](https://learn.adafruit.com/building-circuitpython). I think the steps are:

1. Upgrade the `lib/tinyusb` submodule to the latest commit to get the newest USB Host API. You may need to update the existing calls too.
2. Add calls into [the TinyUSB API](https://github.com/hathach/tinyusb/blob/master/src/host/usbh.h#L147-L168=) into the [shared-module/usb/core/Device.c](https://github.com/adafruit/circuitpython/blob/main/shared-module/usb/core/Device.c#L119-L125=) source file. It is in shared-module because it'll work on all ports that have the `usb_host` module and use TinyUSB.

If you hit any snags, jump on Discord and we'd be happy to try and help.

fadh3r, 2022-06-30:
Thanks, Scott!
I'm done with successful build of custom CircuitPython and now I'm trying to understand architecture of api calls for required changes, and this looks like few sleepless nights))

PR #8155 (2023-07-11): Basic USB host support and keyboard workflow
Connects up read, write and ctrl_transfer to TinyUSB. USB Host support is available on iMX RT and RP2040.

Fixes #6527 (imx) and fixes #5986 (rp2).
commit 2686beab (2023-07-11): Basic USB host support and keyboard workflow

Connects up read, write and ctrl_transfer to TinyUSB. USB Host
support is available on iMX RT and RP2040.

Fixes #6527 (imx) and fixes #5986 (rp2).
closed 2023-07-11 by tannewt