adafruit/circuitpython #6529: "no module named 'traceback'" exception on Matrix Portal M4

tammymakesthings, 2022-06-28:
I'm getting an exception trying to import `asyncio` on my device. I've stripped this down to a minimal test case, which is as follows:

**Device**: Adafruit Matrix Portal M4, running CircuitPython 7.3.1

I wiped the `lib` directory clean on my device and then installed a bare minimum subset of libraries with the command:
```
circup install adafruit_matrixportal asyncio
```
I loaded a minimal `code.py` file onto the device as follows:
```python
import asyncio

print("Hello from asyncio on the MatrixPortal!")
while True:
    pass
```
When the device restarts, the following is logged in the serial console:
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 1, in <module>
  File "asyncio/__init__.py", line 14, in <module>
  File "asyncio/core.py", line 19, in <module>
ImportError: no module named 'traceback'

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
I'm happy to dive deeper into this issue and see if I can figure out what's going on, but I probably won't have time for at least a few days. I haven't tested whether this problem occurs on other boards/platforms yet, or whether it goes away if I down-rev CircuitPython, but I will do those tests and update the issue as soon as I have a bit of time.

FoamyGuy, 2022-07-11:
I confirmed this is resolved now in the `7.3.x` branch. I made a build from that branch for Matrix Portal and tested with the reproducer code from above. It does now successfully complete the import and print the message.

```
Adafruit CircuitPython 7.3.1-2-g5098d0d10-dirty on 2022-07-11; Adafruit Matrix Portal M4 with samd51j19

code.py output:
Hello from asyncio on the MatrixPortal!
```

xiaoyu-tamu, 2022-07-16:
Hi,@FoamyGuy . can u share that build? or is there any guide for build custom circuitpython? 

My product still under development, breaking changes is acceptable.

xiaoyu-tamu, 2022-07-16:
nvm, i found nightly builds in the release page.

dhalbert, 2022-07-17:
I will be making a 7.3.2 pretty soon.

dhalbert, 2022-07-20:
Fixed by #6530.

PR #6530 (2022-06-28): restore traceback to matrixportal build
Fixes #6529.
closed 2022-07-20 by dhalbert