adafruit/circuitpython #6535: Implement TLS for Ethernet

JetForMe, 2022-06-30:
@ladyada said I should open a ticket here. I'd like to work on TLS support for Ethernet (e.g. Wiznet 5500). I have a great deal of general software development experience, but not a ton of Python, and very little knowledge of the micropython/CircuitPython networking stack internals.

If I could get some guidance from someone with more knowledge of the network stack, I could make significant progress.

Ideally, we should be able to make things like TSL/SSL support common to both WiFi and Ethernet (and any other underlying transport).

tannewt, 2022-06-30:
Adding SSL support would be great! You are using the Python Wiznet driver now right?

The MicroPython and CircuitPython network approaches are a bit different now. The CP APIs were designed to support Python-only drivers but MP took a more CPython approach where the OS manages it.

One place to start is the `ssl` module from Python. It is used to wrap sockets. We have a subset of the API here: https://github.com/adafruit/circuitpython/tree/main/shared-bindings/ssl I think we can continue to use this model.

Right now the only implementation we have punts to the ESP-IDF for everything. Having port-agnostic TLS would be awesome though. That'd help in supporting the new Pico W too.

Here's how I'd suggest starting:
1. Join [the Discord chat](https://adafru.it/discord) for #circuitpython-dev where we can help in real-time.
2. Get CP building: https://learn.adafruit.com/building-circuitpython
3. Figure out what sockets we need to wrap and how that impacts the `ssl` module implementation.

victorallume, 2022-10-24:
I'm trying to implement something similar in micropython (on ESP32, or wherever); namely getting SSL working on the pure-python wiznet driver (which was backported from circuitpython). Findings so far: 

- On an ESP32, the SSL library is MBedTLS running on the ESP-IDF. There are micropython and circuitpython bindings to that library, but that have diverged (though there will likely be an SSL context manager coming to micropython soon, so might re-converge at some stage)
- The `socket` class provided by `adafruit_wiznet5k_socket.py` doesn't implement the streaming interface that the SSL module requires; in particular, it needs to inherit from `io.IOBase` and provide methods for `read()`, `write()` and `ioctl()` (the wiznet socket library only provides `recv` and `send`)
- Just implementing the above doesn't work 

While I'm trying to get this going on micropython, I'm happy to share in both directions

nabber00, 2023-06-12:
See #2202

ronpang, 2023-06-20:
Hi Everyone, I'm Ron from WIZnet HK. 

After viewing these issue, I wanted to try to modify our socket section that allows the SSL function could work on our PICO board in circuitpython.

Will there be anyone could help me to give some guideline to work on this development? 

Also, I had join the discord chat as well. Which hastag section that I bring up this topic?

tannewt, 2023-06-20:
> Will there be anyone could help me to give some guideline to work on this development?

I don't know a lot about SSL but can help guide you in CP. @victorallume's comment seems on the right track. The goal would to to have the native `ssl` module wrap a non-ssl, Python-only socket object.

> Also, I had join the discord chat as well. Which hastag section that I bring up this topic?

#circuitpython-dev is the best place to discuss modifying the C core of CircuitPython. Note that most of us are in the US and are unlikely to be online during our night.

ronpang, 2023-06-29:
@tannewt 

Thanks for your explanation. I think I need to study more about the steaming interface to make it happen.

@victorallume 

I found that our HQ members had made SSL/TLS in MQTT for micropython. Maybe it would help.

https://github.com/renakim/W5100S-EVB-Pico-Micropython/tree/main/examples/AWS

https://github.com/renakim/W5100S-EVB-Pico-Micropython/tree/main/examples/Azure

anecdata, 2024-05-17:
Implemented by #8954

closed 2024-05-17 by anecdata