adafruit/circuitpython #6537: Teensy 4.1, problem receiving midi over UART

krixgris, 2022-06-30:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.1 on 2022-06-22; Teensy 4.1 with IMXRT1062DVJ6A
Board ID:teensy41
```

### Code/REPL

```python
import time
import board
import busio
import adafruit_midi

from adafruit_midi.control_change import ControlChange
from adafruit_midi.note_off import NoteOff
from adafruit_midi.note_on import NoteOn
from adafruit_midi.pitch_bend import PitchBend

uart = busio.UART(board.TX2, board.RX2, baudrate=31250, timeout=0.001)  # init UART
midi_in_channel = 1
midi_out_channel = 1
midi = adafruit_midi.MIDI(
	midi_in=uart,
	midi_out=uart,
	in_channel=(midi_in_channel - 1),
	out_channel=(midi_out_channel - 1),
	debug=False,
)

midi_event = None

while True:
	# read whatever the next midi message is
	midi_event = midi.receive()
	# returns None when buffer is empty
	if midi_event is not None:
		# handle NoteOn
		print(midi_event)
		if isinstance(midi_event, NoteOn):
			if(midi_event.velocity>0):
				midi.send(midi_event)
		# handle NoteOff
		if isinstance(midi_event, NoteOff):
			midi.send(midi_event)
```

### Behavior

MIDI Input is not working consistently over UART.
Code prints any midi event that is not "None"
Tried multiple pairs of TX/RX but the same result for all.

### Description

MIDI In does not work over UART. Sometimes MIDI messages land on the input, but 1/30 of the time or even less works.
I noticed sending midi pitch bend or after touch generated *something* on the input more consistently.
Note On/Off seems to mostly not do anything at all.

Testing this by sending MIDI to the input through an adafruit midi featherwing, hooked up to various midi capable devices.
For example Arturia Keystep, as well as USB to midi devices letting me send MIDI with USB from my computer, converted to MIDI DIN.

MIDI Out works as expected on the Teensy though, but MIDI In is erratic at best.

### Additional information

I tried the exact same code, as well as the same MIDI Featherwing on my Adafruit Feather M4 Express and it works perfectly through that.

tannewt, 2022-06-30:
This is likely a UART buffering / buffer size issue.

jepler, 2023-06-28:
@jedgarpark iirc you discovered that setting a UART timeout was required for this to work properly? can you fill in some details? I hope it might be effective to get MIDI over UART working on teensy 4.1 too.

jedgarpark, 2023-06-29:
@jepler i can double check next week, but iirc it was just the `timeout=0.001` argument that needed to be there for it to work.

PR #8928 (2024-02-16): Numerous iMX fixes and SDK update
* Fixes #4954. Leaving GPIO in output mode messed with ADC readings.
* Fixes #8018 and fixes #6537. UART rx was more complicated than it needed to be.
* Fixes #5581. uart reset is handled by finalizer. Added i2c_reset() call and reservation check.
closed 2024-02-16 by jepler