adafruit/circuitpython #6548: web workflow status title line setting with thonny, ampy, etc.

dhalbert, 2022-07-01:
I was trying to use `ampy`, `thonny`, etc. with recent `main`, and discovered that they are not expecting the in-line terminal title-bar terminal escape sequences. `ampy` seems to recover, at least for `ls`, but takes a long time to give results.

I was only testing this with the ESP32 development build, so not sure about the ESP32-S2 build, etc. But I'd expect the same problems.

Neradoc, 2022-07-01:
Similarly, Mu will need an update to make it ignore the control code that creates a title, or do something with it, because it appears on the serial console and I think it causes issues with the REPL.

dhalbert, 2022-07-01:
The title bar should be turned off in the raw REPL, at least.

tannewt, 2022-07-06:
> This is also affecting ESP32-C3 devices, making it painful to bootstrap them into web workflow.
 
We know. It'll be fixed before 8 is stable. For now, please do the `/.env` write through the serial yourself.

tannewt, 2022-07-13:
I've filed an issue on Thonny for guidance: https://github.com/thonny/thonny/issues/2349

tannewt, 2022-08-10:
PR to Mu is here: https://github.com/mu-editor/mu/pull/2326

arturo182, 2022-08-18:
Not sure if this is the right issue for this, but I just flashed the latest main branch, and noticed the title bar is not properly printed in `minicom` too.

I suspect this will be an issue with many serial communication programs.

[image]

dhalbert, 2022-08-18:
It works with `tio` and `picocom`. I am not sure what the issue is with `minicom`. I replicated your issue above. I would think there would be a setting to pass through the codes, but I could not find one.

todbot, 2022-08-18:
Is there a way to turn off sending these escape sequences entirely? Something in `supervisor` or similar?

dhalbert, 2022-08-18:
> Is there a way to turn off sending these escape sequences entirely? Something in `supervisor` or similar?

There is not. There could be a `supervisor` or a `.env` setting. Feel free to open an issue suggest it should be able to be disabled.

tannewt, 2022-08-19:
Both Thonny (https://github.com/thonny/thonny/issues/2369) and Mu (https://github.com/mu-editor/mu/commit/a56049feb68ba8fcad8fa0fd600cc62eb7da8a58) have OSC handling merged in. Thonny 4.0.0b4 includes the changes.

dhalbert, 2022-08-19:
Closing this as completed -- thanks! Further ideas can have their own issues.

closed 2022-08-19 by dhalbert