adafruit/circuitpython #6552: Web workflow on espressif chips appears to get blocked by background FS/Flash operation

RetiredWizard, 2022-07-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-alpha.1-27-g187f156ac-dirty on 2022-07-01; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
```

### Code/REPL

```python
No user code is run on the microcontroller before the behavior occurs.
```

### Behavior

After initiating a GET via curl (and leaving it apparently waiting for a response) I then accessed the CIRCUITPY drive and renamed a file. When the new filename was saved, CURL picked up the following:

     C:\Users\Desktop\curl-7.84.0_3-win64-mingw\bin>curl -v http://10.0.0.189 --output -
     *   Trying 10.0.0.189:80...
     * Connected to 10.0.0.189 (10.0.0.189) port 80 (#0)
     > GET / HTTP/1.1
     > Host: 10.0.0.189
     > User-Agent: curl/7.84.0
     > Accept: */*         <---------------Stops here until FS/Flash operation is completed
     >
     * Mark bundle as not supporting multiuse
     < HTTP/1.1 200 OK
     < Content-Encoding: gzip
     < Content-Length: 482
     < Content-Type: text/html
     <
     Q?~u$?"???????^?????    ??o?+q?PV?§?#?r-fsEe?\~t??"     ? ??P6?,?4z_?¦??1E?8?-Iu?,?i???-???6?
     a?Q&???DS?A?F???????.l?c??Y?6?L?p??,af9U?C????i? \?j'?j?
     ?¤?};*M?h?Jz¦#?p0Yg?!? ?b??9?  * Connection #0 to host 10.0.0.189 left intact?L?L??8??????;???5??f???d?%?f?¦?u>s#?x??B??

When connecting via a browser rather than curl, the board behaved in the same way except that when a flash operation was completed, the web page was served up as expected rather than the curl binary dump shown here.

### Additional information

 I commented out the following block in supervisor/shared/flash.c and the web pages started serving up fine without the need to complete a flash operation first:

    // Turn off ticks now that our filesystem has been flushed.
    if (filesystem_dirty) {
        supervisor_disable_tick();
    }

evildave666, 2022-07-10:
@RetiredWizard built an image for the Lolin C3 Mini, which was exhibiting the same issues with the commented out bits in flash.c and it started working immediately as well.

tannewt, 2022-07-11:
I suspect this is an issue with the main CP task not being woken or the background ticks not being run even when there is socket stuff pending. I'll take a look at this shortly. We may need a separate task to run `select()` on the sockets and then queue up the background task.

RetiredWizard, 2022-07-11:
RTOS is all black magic to me at this point.... In case it's of any use I've been using the web workflow without this block of code and occasionally the workflow still goes away until there is some flash I/O. I've noticed it usually after coming back from an extended period of inactivity.

tannewt, 2022-07-11:
The HTTP code is currently driven by a tick callback. On boards with displays this is works because ticks are on for display refresh. On boards without it, the code can't respond because we're asleep.

I think the fix is to add a `select()` based task that can wake CircuitPython when sockets are ready. We'll also need an eventfd to let the select task know to update the sockets it is waiting for.

PR #6591 (2022-07-15): Add socket select task to wake CP
This fixes the web workflow on boards without a display.

Fixes #6552
commit e9dd2541 (2022-07-15): Add socket select task to wake CP

This fixes the web workflow on boards without a display.

Fixes #6552
closed 2022-07-15 by jepler